mongoose-os-apps/shelly-homekit #723: Power measuring in 2.9.2

kaese2002, 2021-07-09:
Hello

thx for the feature and finally supporting Eve.
But i noticed on a 2.5 it´s only working in switch mode. If you use a 2.5 in roller shutter or garage opener mode the feature isn´t present. Would be could if you can fix that.

thx

rojer, 2021-07-09:
@timoschilling i think it makes sense, can you add it?

kaese2002, 2021-08-09:
Any news on this?

timoschilling, 2021-08-09:
Need to find some time for that

bockph, 2021-09-14:
Hi

> 
> 

> thx for the feature and finally supporting Eve.
> But i noticed on a 2.5 it´s only working in switch mode. If you use a 2.5 in roller shutter or garage opener mode the feature isn´t present. Would be cool if you can fix that.

Hi can you describe how excactly Power measurement is integrated into eve? Or is the corresponding version not yet on the server for OTA update? I Have a 2.5 and 1PM and simply used the links from Readme. Toggling on off via homekit and eve app does work like a charm but no hint wor the PM values (via browser i can see them)

timoschilling, 2021-09-14:
[image]

timoschilling, 2023-01-19:
Implementing it for garage opener would make sense, because garage opener only give a milliseconds impulse.