mongoose-os-apps/shelly-homekit #726: Web is stuck on “Loading” screen after a security password was set

sigito, 2021-07-11:
One of my Shelly 2.5 is stuck on “Loading” screen. This happened after I have set a password to access the web interface. I tried restarting the device, but it had no effect.

When I reset the wifi settings, by following the [Recovery](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery) instructions, the web interface was again accessible. Though once I set the security password again, it returned to the same “Loading” state.

The Shelly itself is operating all good, and I can control it with HomeKit without any issues.

Firmware version: 2.9.2.

Thank you for a great product!

timoschilling, 2021-07-14:
Which browser do you use?

sigito, 2021-07-15:
Initially I did all setup on my iPad running Safari. Though later I tried to use Firefox on macOS and it would still not work.

When I opened a debug in Firefox the console would keep printing regular errors when getting auth. Unfortunately I did not record it earlier, and after today's update to 2.9.3 the issue is gone. Sorry about that.

Thank you!

closed 2021-07-15 by timoschilling