mongoose-os-apps/shelly-homekit #727: Shelly 2.5 does not register simultaneous up-off and down-on presses

sigito, 2021-07-11:
Hi everyone,

A short explanation. Sorry if it is unnecessary.
I have shutters that a controlled by a switch with two button. Only one button at a time can be pressed, so when an “up” button is pressed, the “down” is gets unpressed at the same time, and vise versa. To enter a neutral position a currently unpressed button needs to be pressed but not clicked it.

The issue arises when I toggle directly from one button fully pressed to another, without entering a neutral position. Shelly only registers the un-press event of the button, but does not seem to register that another button was pressed in. Is there a way to go around this issue?

Thank you!

nseries100, 2021-07-27:
Yep, this is frustrating behavior... :/

sigito, 2021-08-28:
@timoschilling, how do I access them? `/debug/log?follow=1` page requires authorization, and using `admin` and my password pair results into a 401 error code.

rojer, 2021-08-28:
> @timoschilling, how do I access them? `/debug/log?follow=1` page requires authorization, and using `admin` and my password pair results into a 401 error code.

yes, debug endpoints are not accessible when auth is on. please disable auth temporarily.

sigito, 2021-08-28:
Scenario: the shades are moving up, I toggle the button to go down.
Result: the shades stop moving.

```
152708920 shelly_hap_window_c:594 Moving to 100, p 98.13
153010103 shelly_hap_window_c:331 WC 1: Cur pos 16.09 -> 16.25, P = 98.35
153509418 shelly_hap_window_c:594 Moving to 100, p 98.13
153810267 shelly_hap_window_c:331 WC 1: Cur pos 17.40 -> 17.57, P = 97.80
154308856 shelly_hap_window_c:594 Moving to 100, p 98.26
154610408 shelly_hap_window_c:331 WC 1: Cur pos 18.72 -> 18.88, P = 98.25
154816444 shelly_input.cpp:76     Input 2: change (state 1)
154823995 shelly_hap_window_c:342 WC 1: Tgt pos 100.00 -> 19.24 (ext)
154831659 shelly_hap_window_c:342 WC 1: Tgt pos 19.24 -> 19.25 (fixup)
154836678 shelly_output.cpp:63    Output 1: on -> off (moving)
154843074 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
154849782 shelly_input.cpp:76     Input 1: change (state 0)
154879090 mgos_sys_config.c:232   Loading conf2.json
155016944 mgos_sys_config.c:174   Saved to conf9.json
155024859 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
155041890 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
155849310 shelly_input.cpp:76     Input 2: long (state 1)
160579667 shelly_main.cpp:543     Up 160.57, HAP 0/5/12 ns 5, RAM: 21180/39048; st 57; 4.1: c:1 mp:98.03 mt_ms:60790 cp:19.25 tp:19.25 lemd:1 lhmd:0
```

rojer, 2021-09-01:
so this is working as expected: pressing down while moving up will stop the movement.
you will have to press down again to get the curtain to start moving down.

sigito, 2021-09-01:
Unfortunately this means I have to press up again to disengage the down button (it is fully pressed in at the end of the scenario), and then press down again. And this is exactly the problem I reported.

marcus-aa, 2021-10-02:
There are many rollers that are controlled by these "tri-stable"-with-center-stop switches and as poster says the action to stop is more logical when button is released from the current direction. Some even has a middle "stop" button that with the current implementation does not work.

[image]

@rojer , would it be possible to detect a long-press and act differently? For example, if UP or DOWN is held for more than 1 second (much longer than a momentary switch normally is pressed), the shelly then goes into flank mode and when the button is released from the long-press, a stop is activated. In both directions of course.

That would give the best user experience regardless of operating bi/tri-stable or momentary switches and keep current behaviour for all momentary actions (<1s)

/M

closed 2021-11-10 by github-actions[bot]