mongoose-os-apps/shelly-homekit #728: Reverse shades movement at the end of an action

sigito, 2021-07-11:
Hi everyone,

My shades can be put in a semi-open state, when the individual panels are turned at a certain angle, between 0 (let the light through) and 90 degrees (fully blocking the light). To do it manually one would roll the shades down first (so all the panels close), and then up for a split second. Is there a way to perform this roll-up action to change the angle after a roll-down was done?

I thought I could do it via a set of steps in HomeKit, but 1% is not granular enough. It opens the shutter plates to a fully horizontal position (allow all the light through), when I would want to have them at 45 degrees for example.

Thank you.

sigito, 2021-07-12:
So this is pretty odd. When for the fully closed shades I manually set them to be 1% open, the panels transition to a fully open state (parallel to the ground). But if I do the same via Automation, they are open by just a bit, which is what I need in the first place.

Taking this into account I don’t think there is much more to do here, except provide some option to open shades by ~1% just after they have been fully closed. Though I suspect this is falling more into a HomeKit domain than in responsibility of this firmware. Feel free to close this bug if this is indeed true.

Thank you.

timoschilling, 2021-07-12:
Sounds like a double of #560

sigito, 2021-07-12:
Yes, you are right, that is exactly what I need :)

closed 2021-07-12 by sigito