mongoose-os-apps/shelly-homekit #729: Loosing Wifi Briefly

5xPa, 2021-07-11:
I have upgraded to version 2.9.2 from stock firmware. Added to Homekit. Device Shelly 2.5
If I connect to the webpage eg 192.168.0.189 occasionally for a very short time "disconnected" message flashes up and the light being controlled switches on and off. 

Settings 
Switch
Input Mode Edge
Initial State  Last
Read through the issues and found reference to 2.9.1 occasionally causing a similar problem. Also found the following command
http://DEVICE/debug/core, issued that command returns "no core dump found" 

It appears to be stable when the web page is not loaded, all control by HomeKit. 
To establish  "Chrome" is the problem?
Used MS "Edge" same result disconnect and light switches on and off.
Don't know if this is relevant I use a Draytek 2927 router and Draytek APs
In the router in has a LAN DNS function which I use eg 192.168.0.189 = diningroom.drayddns.com (set only used for local network)

Browser used is "Chrome" running on Windows 10 Pro
Kaspersky internet security.

rojer, 2021-07-11:
do you see uptime reset? also, if you open a tab with debug log, what does it show?

5xPa, 2021-07-12:
Uptime Resets

Here is a copy of the log
42856328 shelly_debug.cpp:227    No log file, sending new entries
47533796 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.27:52500
48538817 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.27:52500
48609302 shelly_main.cpp:543     Up 48.60, HAP 0/0/12 ns 0, RAM: 19148/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
49533337 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.27:52500
50538489 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.27:52500
51545649 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.27:52500
56609112 shelly_main.cpp:543     Up 56.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
64610337 shelly_main.cpp:543     Up 64.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
72609661 shelly_main.cpp:543     Up 72.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
80609397 shelly_main.cpp:543     Up 80.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
88609705 shelly_main.cpp:543     Up 88.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
96609447 shelly_main.cpp:543     Up 96.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
104609522 shelly_main.cpp:543     Up 104.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
112609356 shelly_main.cpp:543     Up 112.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
120609420 shelly_main.cpp:543     Up 120.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
128609490 shelly_main.cpp:543     Up 128.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
136609693 shelly_main.cpp:543     Up 136.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 65; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
144609470 shelly_main.cpp:543     Up 144.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
152609490 shelly_main.cpp:543     Up 152.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
160609443 shelly_main.cpp:543     Up 160.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
168609567 shelly_main.cpp:543     Up 168.60, HAP 0/0/12 ns 0, RAM: 22340/38376; st 64; 0.1: st:0 in_st:0 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0

5xPa, 2021-07-12:
Rojer, some further info if I close down the web interface, operate via HomeKit. Device can be stable for sometime. If then open the webinterface the device switches off/reboots.
As a test I disabled Kaspersky internet security. Still reboots if I load web interface

5xPa, 2021-07-12:
Hi I think I closed the issue by mistake. Have set to open again.

rojer, 2021-07-12:
ok, i really want to observe the device, so let's try the following:

```
http://A.B.C.D/rpc/Config.Set?config=%7b%22debug%22%3a%7b%22udp_log_addr%22%3a%22144.76.98.107:13001%22%7d%7d&reboot=true
```

substitute your device's ip'address, this will enable sending logs to my server directly (144.76.98.107:13001).

later, to disable it, just set back to empty value:

```
http://A.B.C.D/rpc/Config.Set?config=%7b%22debug%22%3a%7b%22udp_log_addr%22%3a%22%22%7d%7d&reboot=true
```

5xPa, 2021-07-12:
Hi Demoid,
                  Ok info should now be going to your server. I block all
IoT devices from reaching the internet. Have set a rule to to pass the the
Firewall, can you let me know the info is getting to your server?

I set fixed ip address in the shelly app, is there a way of setting or
rather changing  fixed ip in your app. As I would have provided the device
with a new ip outwith the range of ip addresses I block?

How long do you wish me to send the info

rojer, 2021-07-12:
yes, i can see logs from shellyswitch25-018E3E

5xPa, 2021-07-12:
Cheers thanks for the first 5 minutes or so I shall leave the web interface page open, then I shall close the page. As it appears to be more stable if I close the web page. 
approx 13:45 I shall close the page.
Would 1 hours worth of data be enough? So I can re enable Firewall

5xPa, 2021-07-12:
Just to confirm 25-018E3E is the correct device

reh8388, 2021-07-12:
I am seeing a similar issue on at least 1 of my devices after the update this morning. I am not seeing it on the Shelly 1 devices and have not had a chance to check the other 2.5 devices yet to see if it is just the one or all of them.

I can set my unit up to send logs later today if it would be beneficial

rojer, 2021-07-12:
@5xPa i can see device restarts when web interface is used. however, i'm not sure what those are... they are very short, not like anything i'd expect. there are no core dumps, so it's not a crash, but have no idea what else it could be... hm.

later i'll see if i can put together firmware with additional debug for you to try.

@reh8388 yes, please.

5xPa, 2021-07-12:
Thanks for the Update be happy to install an update. Do want to continue to receive log to your server?

rojer, 2021-07-12:
yes, for now

5xPa, 2021-07-12:
Ok I have set a filter to permit that device to access the net

rojer, 2021-07-12:
ok, here's debug firmware: http://rojer.me/files/shelly/2.9.2-dbg1/

it is modified to dump core on all intentional restarts. if firmware decides to reboot, we should know from dump why.
please install and report if you get anything in /debug/core.
memory dump contains wifi password in plain text so please don't share publicly, or set up a temporary network.

reh8388, 2021-07-12:
I've made it back to the devices and the one unit that was exhibiting this behavior this morning appears to have stabilized. The WebUI loads without issue and the device has had an uptime of 10+ hours now. I will continue hitting the webUI and see if it reproduces.

5xPa, 2021-07-13:
I have noticed if the device is "off" when WebUI loads it is much more stable. If the device is "on" switched from iPhone and open WebUI almost everytime the device switches "off" reboots I think.

5xPa, 2021-07-13:
Downloaded and installed debug firmware. Tested as follows

iPhone/iPad can control device.
Device switched "On" from iPad, the try to load WebUI device switches "off" page takes a longtime to load, device switching "on and off" rebooting?
Close web page access wait a few minutes switch device "off" from iPad.
Load WebUI Ok
Issue http://192.168.0.189/debug/core    result "no core dump found"

5xPa, 2021-07-13:
correction it replies "no core dump" I wrote in previous comment "no core dump found"

5xPa, 2021-07-13:
Demoid,
            Kept trying and eventually extracted a core dump! How do I send
to you? Taking your advise not to send publicly

rojer, 2021-07-13:
rojer@rojer.me

rojer, 2021-07-13:
got it. but no, core dump is the result of disabling udp logging - you were running /rpc/Config.Set udp_log_addr="".

so, based on what you said, device reboots involuntarily.
based on what you said about device being less stable when the switch is on, i suspect it's some sort of voltage instability.
loading web ui involves a lot of processing on the device, and when UI is loaded it updates the status every second.
it must be some sort of hardware issue.

5xPa, 2021-07-13:
I have 6 2.5s and 6 shelly 1 they seem to be stable. When I run in stock
firmware never noticed an issue. Though to be fair it sometimes took awhile
to load. I shall get another device and install. Thanks for the assistance.
I shall revert back to stock firmware, on the suspect device.

patricks, 2021-07-15:
Since firmware version `2.9.2` and with `2.9.3` I have similar issues. But also only on one Shelly 2.5. This one has the poorest Wifi Signal. But the previous firmware version didn't reboot constantly.

5xPa, 2021-07-15:
Have updated to 2.9.3, the Shelly 2.5 was not the best WiFi speed, checked in DrayTek AP it shows the TX and RX speeds. So far on 2.9.3 seems stable. The unit that was rebooting I revert to stock firmawre. To check if it was a hardware issue. 24 hours and has been stable

closed 2021-08-22 by github-actions[bot]