mongoose-os-apps/shelly-homekit #734: Shelly 1PM new firmware 2.9.2 update issue

Tony240111, 2021-07-13:
Today I update HomeKit firmware from 2.9.1 in to 2.9.2 and now my Shelly 1PM has big issue. 
After update to new firmware shelly completely disconnect and is not possible to connect with any app to remote switch. connection is completely lost.
And also lights when switch getting disconnected are not possible to turned them on neither manualy on actual switch. 
Only option to bring Shelly 1PM back to life is completely turn off and back on power from main power board.

I need to help how to fixed this issue ASAP please ???

rojer, 2021-07-13:
ok, so powering shelly off and back on brings it back? does it connect to wireless? how long does it stay up after that?

Tony240111, 2021-07-13:
1/2 hours - 1hour  sometime less but never longer than 1 hour

Tony240111, 2021-07-13:
firmware 2.9.1 has sometime some problems but worked much better as 2.9.2

I need to fixed this problem ASAP Please

rojer, 2021-07-13:
you can downgrade to 2.9.1 by downloading it from here - http://rojer.me/files/shelly/2.9.1/
then use file upload for in the firmware section.

but before that, let's see if can figure out what the problem is.
to help with that, please open the "log" link and leave it open. let's see if something specific happens before device goes offline.

bmpalmeida, 2021-07-13:
In one of my devices that i have exactly this problem, i manage to get this log. don't know if related, but it is on the device that is having this behaviour.

15223660 shelly_debug.cpp:227 No log file, sending new entries
16393666 esp_main.c:138 SDK: pm open,type:0 0
16482605 shelly_main.cpp:543 Up 16.47, HAP 0/0/12 ns 0, RAM: 23572/38600; st 45; 0.1: st:0 in_st:0 inm:1 ininv:0
18811042 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
19974133 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
21201798 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
21818638 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
22522692 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{�&•…€ó.�•' 00000000 00000000
22819433 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
23941485 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
24482538 shelly_main.cpp:543 Up 24.47, HAP 0/0/12 ns 0, RAM: 25468/38600; st 44; 0.1: st:0 in_st:0 inm:1 ininv:0
24813496 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
25814772 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
26296024 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{H!(�' 00000000 00000000
26831422 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
27891400 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
28930531 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
29817057 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
30186118 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{†� �' 00000000 00000000
32482357 shelly_main.cpp:543 Up 32.47, HAP 0/0/12 ns 0, RAM: 25204/38600; st 45; 0.1: st:0 in_st:0 inm:1 ininv:0
32569330 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
33582900 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361
34580969 mg_rpc.c:305 Shelly.GetInfoExt via WS_in 192.168.50.96:52361

rojer, 2021-07-13:
looks like some noise on UART lines. shouldn't be a problem in itself, but can you try downgrading to 2.9.1. is it more stable? do you see this noise in the logs?

Tony240111, 2021-07-14:
I have 4 Shelly 1PM relays. 2 Relays I updated with 2.9.2 and this 2 Relays has the same problem with disconnecting power as I comment before and another 2 with no updated firmware 2.9.1 works fine. what benefit new update have?

rojer, 2021-07-14:
ok, my apologies, i think 2.9.2 shipped with a bug that may have cause network instability. i pushed 2.9.3, please update.

Tony240111, 2021-07-14:
You think 2.9.3 now will work good with no problem?
And I have question about new updated version 2.9.3 before I will do update: what this update will fix change from version 2.9.1 

Thanks

rojer, 2021-07-14:
yes, i think 2.9.3 is better than 2.9.2
changes in 2.9.2 are described [here](https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.9.2)
connection stability was actually the goal, but looks like a bug has crept in.

killi23, 2021-07-14:
Hi. I have 2 Shelly  1's. Both were on 2.9.2 and dropped from network ,unresponsive in home app every 4-5 toggles until power cycled. 1 updated through the Shelly homekit UI to 2.9.3  but it still has issues like before ,the 2nd refuses to update the same way. When I try to downgrade to 2.9.1 "the update from file" the screen refreshes and the file that im trying to upgrade to 2.9.1 disappears. Any help would be appreciated.

Tony240111, 2021-07-14:
I have the same problem with my shelly 1PM 2 adapters I have updated in to new firmware 2.9.2 and now not possible to fix another 2 I have not updated yet still run on 2.9.1 works ok once new stable update relishes again I will update again. Is too much work for me to try, but for this 2 already updated shelly's, I really need help to fixed them back in to normal working condition please

Thank You

Tony240111, 2021-07-15:
Hi There,
Yesterday I updated my shelly 1PM with new firmware 2.9.3 I don't know why but this new firmware didn't fix my problem from last update 2.9.2 I reported last time. 
Situation is still the same shelly getting lost from power and only option to bring him back is turn off and on main switch in meter box. So painful this is!!! Looks is no way this problem can be fixed???

rojer, 2021-07-15:
very weird. we'll need logs from before the device comes back online.
to get them, let's turn on file logging:
```
http://A.B.C.D/rpc/Config.Set?config=%7b%22file_logger%22%3a%7b%22enable%22%3atrue%7d%7d&reboot=true
```
substitute the address of the shelly for A.B.C.D. you should see `{"saved": true}` and device will reboot.
now, wait for it to become unreachable, power-cycle and immediately after it comes back online open the device log ("Log" link in the system section of the UI). save contents to a file and attach here or paste somewhere (e.g. https://gist.github.com).

killi23, 2021-07-16:
How do I submit logs to you?

rojer, 2021-07-16:
@killi23 you can email them to rojer@rojer.me

Tony240111, 2021-07-21:
I bought Shelly with reason because I owned most apple devices  and HomeKit is very handy for my need. With HomeKit firmware upgrade Shelly 1PM I need to turn off and back on main power in meter box at list twice a day for the reason: Device was lost from power connection. Two days ago I gave up on HomeKit with shelly 1PM and revert firmware back to stock. Thank God this was easy and successful. In last two days testing in stock firmware the same device Shelly 1PM works perfect. But only problem now is in Stock, there is no HomeKit option. I am fully disappointed with all this company, my wasting time to fixed something what is not easy and also dangerous if not working as should work. (lost connection sometime overheat...) I decided to sell all Shelly devices and replaced with another brand ones. My opinion and new experience about HomeKit if smart devices we build in so expensive house to run them with so expensive devices as apple are not originally supported with HomeKit, than better and safer not build them in.

5xPa, 2021-07-21:
There is an option for stock shelly devices operated by HomeKit. It requires an RaspBerry PI running a HomeBridge Shelly plugin

Tony240111, 2021-07-21:
No Thank You. I have better option and final decision. I will sell them all. I have no time wasted my time and also I like my house I don't like make any fire sorry to ask for help. From now on only original factory licensed devices with HomeKit will work in my house. Life is too expensive to do mistakes all the time and try something what maybe works or maybe not.

closed 2021-07-31 by rojer