mongoose-os-apps/shelly-homekit #736: Downgrade version to 2.9.1

killi23, 2021-07-15:
Hi . Looking for a bit of help as still having issue with 2.9.3 and have another shelly running 2.9.1 which i'm not having any issues with. When I try to downgrade in the UI with the 2.9.1 zip  and hit upload the screen just refreshes itself. Is there another way to downgrade the shelly ? .Any help appreciated .

timoschilling, 2021-07-15:
If you are on a Mac or Linux system you can do this command. 
`curl -F commit_timeout=120 -F file=@build_Shelly25/fw.zip http://shellyswitch25-B955B6.local/update`
Change the path after the @ to you local file and adjust the host name. 
Or if you have python installed you can youse the script located unter tools in this repo

killi23, 2021-07-15:
Thanks for the reply . Yes on a Mac.I'm not very good with command line / terminal but I could manage. You said adjust the host name ?
Thank you

timoschilling, 2021-07-15:
shellyswitch25-B955B6.local must be the host name or ip of your shelly

killi23, 2021-07-16:
Perfect . Thank you. I was able to apply 2.9.1 . Very much appreciated.

rojer, 2021-07-16:
please let me know if going back to 2.9.1 made the device stable again.

killi23, 2021-07-16:
Hi . As soon as I applied 2.9.1 it reverted to 2.9.3 . Is there a way to stop it from auto updating ? Thank you

rojer, 2021-07-16:
there is no such thing as auto-updating. it will show the "update" badge but as long as you don't press the button, it won't be applied.

killi23, 2021-07-16:
when I install or revert to 2.9.1 the shelly reboots and shows me 2.9.1. Within 2 minutes the UI shows me 2.9.3.I have 2.9.1 running on another shelly I downgraded without any issue. Both shelly are about 5 meters apart .

rojer, 2021-07-16:
ah, that's because of the commit_timeout=120 in the command Timo gave.
if not explicitly committed within 2 minutes, it rolls back. it's a safety mechanism we use in development and testing, but you don't need it. so just remove it from the command:

```
curl -F file=@build_Shelly25/fw.zip http://XXX/update
```

after this it should stay at 2.9.1

killi23, 2021-07-16:
Thank you for all your help. Not in front of my Mac at the moment . Will apply what you said and update you. Thank you again .

killi23, 2021-07-16:
hi . I applied your code curl -F file=@build_Shelly25/fw.zip http://XXX/update. After 2 minutes the shelly still reverts to 2.9.3.:(
curl -F file=@/Users/xxxxx/Desktop/SHELLY\ 2.9.1/shelly-homekit-Shelly1.zip  http://xxxxxxxxxxxx/update
Is there anything else I can do? 

Thank you

rojer, 2021-07-16:
ok, after the device boots into 2.9.1 after update, go to `http://XXX/update/commit`. this should make it stick.

killi23, 2021-07-16:
Perfect thank you once again . Works perfectly .In relation to the 2.9.1 update I think it suits my environment better. As I said its been rock solid .I have one behind a bathroom mirror and its been flawless and just testing this new one with a bathroom extractor fan on 2.9.1 , im not getting any "no reponse " in the home app after numerous on/off tests. It could be just my network set up. Thank you once again .

closed 2021-07-16 by killi23

rojer, 2021-07-16:
great. i'd really like to get to the bottom of your issues with 2.9.1 though, i'm surprised

rojer, 2021-07-16:
if you don't mind, we could continue on #734.
i need logs from the device.

killi23, 2021-07-16:
How do I submit logs? I have no issues with 2.9.1 . None at all .

rojer, 2021-07-16:
go to 2.9.3, capture the logs when you are having issues, submit them, then go back to 2.9.1.
use the "Log" link in the debug section, turn on file logging as described in recent comment on #734