mongoose-os-apps/shelly-homekit #737: Motion Sensor Activation

victor-vz, 2021-07-17:
Unfortunately I can not run the Shelly 1 as a motion sensor in detached mode as the activation via Homekit takes too long. (The light activates when im already through the stairwell).

Therefore I run the PIR sensor and the light as an output (HAP Type) with activation (Input Mode) and Auto Off Delay. This works perfectly fine, however the Motion Sensor is not anymore exposed to HomeKit which I would like to use for other Automations.

Would it be possible to add an „Manual Activation“ toggle + Auto Off Delay, when using an Outlet in Detached Mode with an Motion Sensor as Input? This would make it possible to instantly turn on the lights and still have the motion sensor exposed to Homekit.

Hope you understand what I am trying to say. As always great work!!! And looking forward to your answers

lucaspinelli85, 2021-07-19:
In mode activation, the auto off start counter when input go to off, it keep on for auto off time, different by other mode that the auto off start counter when input go on.
So, in activation mode the time of on is: time input is on + time auto off.
In other mode the time of on is the time auto off.

victor-vz, 2021-07-19:
That is correct and totally fine. However I would like the motion sensor to be exposed in Homekit (only possible in detached mode) AND have the motion sensor trigger the lights directly (possible in activation mode). 

Triggering the lights via the motion sensor (detached mode) in Homekit is not fast enough. Therefore I suggested to have an option available when motion sensor is choosen as Input 1 that the shelly can still trigger the light directly. Then lights would turn on instantly but the motion sensor would still be exposed to Homekit for other automatisations.

rojer, 2021-07-19:
@victor-vz what you are requesting calls for more advanced internal architecture than we currently have.
the request makes sense though, i will keep it in mind when doing re-plumbing work.
i cannot promise it will be done soon though.

victor-vz, 2021-12-27:
@rojer Any updates or a rough time horizon when this could happen?

rojer, 2021-12-27:
no update for now