mongoose-os-apps/shelly-homekit #739: Problems adding Shelly to HomeKit 2.9.3

BlancoDeMallloca, 2021-07-18:
Hi all,

wanted to give you a feedback on the new  HomeKit QR Setup. 

Feature is great 👍 

I was quite struggling when trying add two Shelly 1 to HomeKit. I was trying to add the devices for approx 1 hour with no success.

The trick to get it work was to open the log file. This made the HAP more stable and I could finally add both Shelly 1 to my HomeKit. 

Don‘t know if this is only a problem that affects me or if this is a known issue. 

Thanks for the great work within this project.

BlancoDeMallloca, 2021-07-18:
Additional information
May this is because the RSSI was quite high (above 70) and therefore HAP could not get a stable connection.

rojer, 2021-07-18:
> The trick to get it work was to open the log file. This made the HAP more stable and I could finally add both Shelly 1 to my HomeKit.

hm, weird. the only way that could be related is by providing constant traffic from the device. although eb ui sens status requests every second, that should be enough...
ok, i don't see how this could be related. nevertheless, thanks for reporting.

rojer, 2021-08-18:
i believe this is fixed in latest master

closed 2021-08-18 by rojer