mongoose-os-apps/shelly-homekit #740: Siri won’t turn off shelly 1

Chpedro93, 2021-07-22:
Hi everyone,

I’m facing some issues regarding my shelly 1’s that controlling my lights. 
Until this last update I was able to tell Siri to turn on or off the light.
However now it only recognise the turn on command. If I tell to turn off the light it says that scene wasn't setup yet.
Any clues? Thanks

rojer, 2021-07-22:
try removing and re-adding device to homekit. devices have no awareness of scenes, this is purely controller-side concept. looks like something went awry there.

5xPa, 2021-07-22:
I had a similar issue I could issue "On" from Homekit, "Off" did not function. Going into setup>>wifi I could see the shelly WiFi name clicked on it, setup WiFi name and wifi password for local network. It then worked Ok, both On and Off from homekit. Note this was after reverting to stock firmware
What I don't understand how shelly was accepting commands from HomeKit!

closed 2021-08-29 by github-actions[bot]