mongoose-os-apps/shelly-homekit #741: Shelly I3 as Doorbell is not getting camera snapshot on homekit Doorbell events

appliko, 2021-07-23:
Hi,
I'm not sure this issue is related to Shelly/Mongoose, but I have to start from somewhere

In same homekit room, I have a camera (ffmpeg plugin from homebridge) with stream and single frame support, and a doorbell (Shelly I3 with first input as doorbel, other 2 disabled)

If someone ring at the door, I get the event but the snapshot from the camera is not attached
In my previous setup, where I was using the I3 as a motion sensor to trigger a doorbell virtual switch from the camera through an automation, it was working
Also when I was having Nello, camera was picked up automatically
Any idea ? Do you think it could be Doorbell related or more Homekit in general ?

timoschilling, 2021-07-23:
At the moment HomeKit shows only the video if the doorbell it self is the video source. So this is HomeKit related

closed 2021-07-23 by timoschilling

appliko, 2021-07-23:
Got it. WIll try as motion sensor or occupancy, to get notifications with image