mongoose-os-apps/shelly-homekit #743: REST API call URL on SW change

jsiegenthaler, 2021-07-25:
Hi

I have a Shelly1 on original Shelly firmware, and others on your fantastic native Homekit mongoose firmware

In one of my applications of the shelly1, it operates in detached mode.

The SW input calls a URL when the SW input is triggered. 
Any chance of adding this to the your firmware?

Ideally I could enter a URL of my choice and it is called (with a GET) when the SW is activated.

timoschilling, 2021-07-26:
as the readme says it is not supported

jsiegenthaler, 2021-07-26:
thank you, however the readme says not **yet** supported

My ticket here is to show that I wish it to be supported at some time in the future

andyblac, 2021-07-29:
I think this is a duplicate of https://github.com/mongoose-os-apps/shelly-homekit/issues/147, reopen if you think it is not and explain why.

closed 2021-07-29 by andyblac