mongoose-os-apps/shelly-homekit #744: Shellyi3 - problem to revert

slaw47, 2021-07-26:
When I try to revert is come message:
Error downloading: TypeError: Origin http://192.168.68.78 is not allowed by Access-Control-Allow-Origin

The original file also not exist.

rojer, 2021-07-26:
can you attach a screenshot? i'm not sure where you can see that message

slaw47, 2021-07-26:
Hello,

Please find it under Revert button

Pozdrawiam / Best regards
Slawomir Wilinski

slaw47, 2021-07-26:
Hello,

When I try download the Stock firmware from the official URL's below, and upload via web interface.:

I received the screen like print screed.

Pozdrawiam / Best regards
Slawomir Wilinski

rojer, 2021-07-26:
i don't see any images. i think you are responding by email and github doesn't add attachments.

bikejoco, 2021-07-26:
Hi,

I also will revert to the original Shelly Firmware. But I get the following Error:
[image]
How can I fix this Problem? In the Router-Settings (FritzBox 7530) the Shelly is allowed to connect to the internet.

Thanks for Help

JOHANNES

slaw47, 2021-07-26:
Please find attachments: 
[image]
[image]

rojer, 2021-07-26:
firmware is actually fetched by your browser and then uploaded to shelly. the browser must be able to fetch `https://rojer.me/files/shelly/stock/XXX.zip` (XXX being the stock model) and then upload to the device.
what browser are you using?

rojer, 2021-07-26:
hm, offical API endpoint does not provide firmware for shelly i3, i don't see SHIX3-1 in the JSON document. weird.

slaw47, 2021-07-26:
Safari 14.1.2 (15611.3.10.1.5, 15611)

bikejoco, 2021-07-26:
I use Firefox 89.0.2 on MacOS

slaw47, 2021-07-26:
the other one Shelly Switch1 I can to download, the Shelly i3 not

rojer, 2021-07-26:
right, there's no official stock fw for i3 for some reason, i am investigating.

bikejoco, 2021-07-26:
> right, there's no official stock fw for i3 for some reason, i am investigating.

THANKS

rojer, 2021-07-26:
i found 1.10.4 in my archive and put it up. you should be able to revert now.

rojer, 2021-07-26:
i believe this is fixed, feel free to reopen if not.

closed 2021-07-26 by rojer

slaw47, 2021-07-26:
All is OK, thank you