mongoose-os-apps/shelly-homekit #745: Same problem shelly1-homekit #539

SanchoCZ, 2021-07-27:
Hi, 
I have the same problem. Deleting Homekit didn't help. Can there be something in the Apple TV settings? Other certified accessories are displayed on both Wi-Fi and LTE.

App: switch1 2.9.3 20210714-205744/2.9.3-g2f89855-master
Uptime: 5330.76
RAM: 24244 free, 14856 min free
HAP server port: 9000
HAP config number: 103
HAP connection stats: 0/2/12
HAP connections:
  192.168.1.156:49354 nc 0x3fff5a94f 0x200000 io 7 ts 0x3fff51ac rd 6
  192.168.1.204:60563 nc 0x3fff52e4f 0x200000 io 7 ts 0x3fff5b8c rd 2487
 Total: 2
HAP sessions:
  0: s 0x3ffef4d0 ts 0x3fff5b8c o 1 st 1 ts 5324315
  1: s 0x3ffef810 ts 0x3fff51ac o 1 st 1 ts 5324192
 Total: 2

192.168.1.204 is iMac IP

timoschilling, 2021-07-27:
Do you have an HomePod, Apple TV or iPad at your home, which is setup as home hub?

timoschilling, 2021-07-27:
I think your problem is in your HomeKit setup and not on your Shelly. The Shelly has no knowledge about outside WiFi access, that’s only done by the home hub. 

Possible other problem we found in a similar issue: if a other accessory or bridge response to slow it can be happen that a other accessory (the Shelly) will not be updated 

I close this issue because it’s not a Shelly issue

closed 2021-07-27 by timoschilling

SanchoCZ, 2021-07-28:
I have an Apple TV. What can be in the HomeKit settings if I control other accessories (vocolinc) outside the home network?Why don't they just see Shelly's devices?

timoschilling, 2021-07-28:
For example your Apple TV didn’t have a connection to your Shelly only your iPhone and your Mac have one

SanchoCZ, 2021-07-28:
should the Apple TV IP address appear in that log?

andyblac, 2021-07-28:
Yes the AppleTV hub IP should be listed

SanchoCZ, 2021-07-29:
I have this log when I am logged in via LTE

App: switch1 2.9.3 20210714-205744/2.9.3-g2f89855-master
Uptime: 133658.28
RAM: 24452 free, 14160 min free
HAP server port: 9007
HAP config number: 7
HAP connection stats: 0/0/12
HAP connections:
 Total: 0
HAP sessions:
 Total: 0

SanchoCZ, 2021-07-29:
[image]

SanchoCZ, 2021-07-29:
[image]

SanchoCZ, 2021-07-29:
I put Apple TV in the factory settings, I removed the Home application. He set everything up again. The result is the same :-(

timoschilling, 2021-07-29:
Maybe you didn’t setup you Apple TV as home hub. If your Apple TV isn’t connected to the Shelly that’s is definitely the problem. The Apple TV acts as proxy to your home. Read the apple support pages about how to setup.

timoschilling, 2021-07-29:
About your VOCO device: is it WiFi or Bluetooth based?

SanchoCZ, 2021-07-29:
WiFi
https://www.vocolinc.eu/support/english-tutorials/vocolinc-powerstrip-vp3

SanchoCZ, 2021-07-29:
Is this right?
[image]

andyblac, 2021-07-29:
yes, as this firmware dose not have the official license from Apple.

andyblac, 2021-07-29:
make sure you have HomeKit support enabled on your AppleTV.

SanchoCZ, 2021-07-29:
This settings? 

[image]

My Home is connected

SanchoCZ, 2021-07-29:
I was connected via cable.  when I disconnected the cable, I entered the data for wifi and it works.  you also don't have a cable connected?  I have a dual band modem