mongoose-os-apps/shelly-homekit #747: Shelly 1L 'hangs' after being on for ~1 hour

mvnixon, 2021-07-29:
Noticed a problem with my 1L. When activated via HomeKit, after approx 1 hour, I noticed the lights are off. It disconnects from Wi-Fi, and I can't access it unless I flip the breaker to restore power.

I managed to get a log dump as it was dying:

`3840633362 esp_esf_buf_monitor.:75 TX is stuck!
3840637754 mgos_system.c:58        Rebooting in 1000 ms
3840644893 shelly_main.cpp:365     HAP server state: 2
3840655108 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff2eb4 nc 0x3fff63fc 0/8/12
3840663790 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff744c nc 0x3fff747c 0/7/12
3840671606 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5b84 nc 0x3fff6e2c 0/6/12
3840679800 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6c64 nc 0x3fff6f6c 0/5/12
3840688038 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff708c nc 0x3fff6704 0/4/12
3840696146 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c14 nc 0x3fff689c 0/3/12
3840704339 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff8e24 nc 0x3fff717c 0/2/12
3840712149 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6e04 nc 0x3fff738c 0/1/12
3840720070 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6844 nc 0x3fff7534 0/0/12
3840734067 HAPPlatformTCPStrea:105 0x3fff54ec 0.0.0.0:9000 HAP connection closed, ns 0/0/12 i 1
3840741345 shelly_main.cpp:365     HAP server state: 0`

So it looks like it reboots, but never comes 'back'. Thoughts?

rojer, 2021-07-29:
can you try 2.9.4-beta? http://rojer.me/files/shelly/2.9.4-beta1/
a fix for stuck TX queue was commited to the espressif SDK, 2.9.4 is built with tthat fix.

mvnixon, 2021-07-29:
Strange - try the /ota method and get a 'not found' error, try manually uploading and get the following error:

`123786737 shelly_main.cpp:804     Starting firmware update
123798453 esp_ota_backend.c:185   Slot 0, FW: switch1l.bin -> 0x8000, FS fs.bin -> 0xbb000
123810206 mgos_ota_core.c:505     0.18% total, esp_init_data_default_v08.bin 128 of 128
123924311 esp_ota_backend.c:338   Start writing fs.bin (262144) @ 0xbb000
123960603 mgos_ota_core.c:505     0.24% total, fs.bin 512 of 262144
125140662 shelly_main.cpp:543     Up 125.13, HAP 0/0/12 ns 0, RAM: 19596/38592; st 34; disabled
128788113 mgos_ota_core.c:818     Update finished, result -13 (Unexpected end of data)`

rojer, 2021-07-29:
/ota?url method only works on stock firmware. HK firmware has the upload form, you may need to try a couple times

mvnixon, 2021-07-29:
repeatedly get the following error:
`89854692 mgos_ota_core.c:818     Update finished, result -13 (Unexpected end of data)`

Tried different browsers, downloading new copies, no luck. Any ideas?

mvnixon, 2021-07-29:
Never mind - both Firefox and Safari couldn't upload correctly. Chrome saved the day! I'll reply back when I observe the issue not occurring anymore.

rojer, 2021-07-30:
hm, you are right, upload doesn't work in firefox. thanks for reporting, i'll have a look

timoschilling, 2021-07-30:
I can confirm it didn't work in Safari as well.

mvnixon, 2021-07-30:
Apart from the web browser upload bugs, looks like the beta firmware is the solution to that issue. Thanks!

rojer, 2021-07-31:
fixed in 2.9.4

closed 2021-07-31 by rojer