mongoose-os-apps/shelly-homekit #748: Problems adding Shelly to HomeKit - Firmware version 2.9.3

hluhring, 2021-07-29:
When I try adding my first Shelly 1 to HomeKit, I get the message "Unable to Add Accessory -- Failed to add the accessory."  I'm using the Home app on my iPhone. Both the iPhone and Shelly 1 are on a 2.4 GHz Wifi network on a Verizon G3100 router. There is no mesh wifi network involved. 

On the web UI, in the HomeKit Settings section, I click on Setup. This gives me a QR code. I aim my iPhone camera at this. It brings up the Home app. I click "Open in Home". It shows that it is a switch. I click "Add to Home". I choose my home. It says "Uncertified Accessory". I click "Add Anyway". It says "Connecting to Switch". 

And now it works!!  It didn't used to!!  I tried it several times, and it would not add the accessory. Just before composing this message, I rebooted the Shelly 1. And now it works. I'll go ahead and submit this in case it helps someone else.

Xaruqui, 2021-08-02:
Hi there, happened exactly the same to me. Wouldn't connect to HK until rebooting.

Thanks for sharing.

rojer, 2021-08-02:
is this with 2.9.5 or still 2.9.3?

Xaruqui, 2021-08-02:
Sorry I didn’t mention it. Latest 2.9.5

Congratulations for your work, I fucking love it!

rojer, 2021-08-02:
setting up pairing involves some expensive operations that sometimes cause the device to miss a few AP beacons and wifi connection to get dropped as a result, may need a couple tries before it's successful.

hluhring, 2021-08-03:
> is this with 2.9.5 or still 2.9.3?
> […](#)

2.9.3

closed 2021-08-04 by hluhring

rojer, 2021-08-08:
after some extensive work, i managed to reduce the amount of time device is unreponsive during pairing setup and improved pairing success rate significantly, at least in my setting. it still occasionally fails but it's much better tahn before. this will be released in the next version.

Xaruqui, 2021-08-08:
Many thanks. This is highly appreciated.