mongoose-os-apps/shelly-homekit #749: Shelly Plug US shows up as shelly-plug-u1, "will not update shelly-plug-u1 with shelly-plug2".

Numbski, 2021-07-29:
```
103016521 shelly_update.c:168     Starting ota from http://rojer.me/files/shelly/shelly-homekit-ShellyPlug.zip
103521936 mgos_ota_http_clien:270 Update URL: http://rojer.me/files/shelly/shelly-homekit-ShellyPlug.zip
104063910 mgos_ota_core.c:253     Starting, timeout 240, commit timeout 0, mem 36244
104075316 mgos_ota_core.c:487     FW: shelly-plug2 esp8266 2.9.3 20210714-210002/2.9.3-g2f89855-master
104080117 shelly_update.c:90      BEGIN: shelly-plug2
104085763 shelly_update.c:98      will not update shelly-plug-u1 with shelly-plug2
```

I *could* override this manually and force it to go, as I seriously doubt the esp8266 or esp32 inside is wired differently, and I do have the ability to hot-air it off and manually flash it back to something usable, but I thought I'd ask about it prior to attempting that!

Numbski, 2021-07-29:
As it turns out, it's a good thing I didn't!  I was completely wrong - the GPIOs are used differently between the plug and the plug US.

https://templates.blakadder.com/shelly_plug_US.html
and
https://templates.blakadder.com/shelly_plug.html

GPIO # | Component
-- | --
GPIO00 | Led1
GPIO01 | None
GPIO02 | Led2i
GPIO03 | None
GPIO04 | Relay1
GPIO05 | BL0937 CF
GPIO09 | None
GPIO10 | None
GPIO12 | HLWBL SELi
GPIO13 | Button1
GPIO14 | HLWBL CF1
GPIO15 | LedLink
GPIO16 | None
FLAG | None

Versus the plug:

GPIO # | Component
-- | --
GPIO00 | None
GPIO01 | None
GPIO02 | None
GPIO03 | None
GPIO04 | Relay 1
GPIO05 | HLW8012 CF
GPIO09 | None
GPIO10 | None
GPIO12 | Button_i 1
GPIO13 | Led 1
GPIO14 | Led 2
GPIO15 | None
GPIO16 | Led 3
GPIO17 | None

timoschilling, 2021-07-30:
The reason why it doesn't flash is that the US plug has a different name then the (EU) plug2, shelly-plug-u1 vs shelly-plug2. During the update the name of the current and the new firmware is compared and the flash will be aboard if not match.

I think you mixed some things: There are 3 Shelly plugs out there
1. Shelly Plug
2. Shelly Plug 2
3. Shelly Plug US

The pinout you listed as "Shelly Plug (shelly-plug-2)" is that one of the "Shelly Plug", but the "Shelly Plug US" is more like the "Shelly Plug2".

I can give you a test build for the Plug US, but I have no test hardware at hand so I need to trust the Tasmota template, so it can happen that it brick your device.

Numbski, 2021-08-01:
I have 4 sitting here and the ability to pop it open and manually re-flash, so let’s do it. :)

Also, I did paste froM the tasmota template for us, but I was initially confused.

rojer, 2021-08-02:
@Numbski ok, try this one - http://rojer.me/files/shelly/shplgu1.zip
built from the commit above, i think it might work.
if it does, please report if power consumption figures look ok (a good way to test is pluggin a heater, or a a fan, or iron and comparing power consumption numbers).
also, let me know if system temperature in web ui looks reasonable (should be 45-60C or thereabouts).

Numbski, 2021-08-03:
Thank you for that.  My mother-in-law is critically ill, and we had to leave on a moments’ notice to get to her, and my return timeline is unknown.  I will get to this when I can.  I am so sorry for you to go to that effort then make you wait.  I may be able to do it remotely, but that may not be feasible.

Tony Shadwick

rojer, 2021-08-03:
@Numbski oh, sorry to hear that! don't worry, there's absolutely no rush with this.

ryanshuck, 2021-08-04:
> @Numbski ok, try this one - http://rojer.me/files/shelly/shplgu1.zip
> built from the commit above, i think it might work.
> if it does, please report if power consumption figures look ok (a good way to test is pluggin a heater, or a a fan, or iron and comparing power consumption numbers).
> also, let me know if system temperature in web ui looks reasonable (should be 45-60C or thereabouts).
> also please test reverting back to stock.

I was able to try the OTA update with this build, but it doesn't take for me.

timoschilling, 2021-09-04:
@Numbski any updates with your testing?

Numbski, 2021-09-05:
Not yet.  I am back in town though, will try tomorrow.

Tony Shadwick

just-inlewis, 2022-07-19:
I'd be willing to test a new firmware on my plug if needed. It appears the previous iteration didn't work?

just-inlewis, 2022-07-21:
I tried compiling @rojer 's commit after rebasing master but the firmware doesn't apply for me either.
`{"ts":1658371765.154, "level":2, "data":"mgos_ota_core.c:269     Starting, timeout 600, commit timeout 120, mem 159140\n"}
{"ts":1658371765.180, "level":2, "data":"mgos_ota_core.c:656     FW: shelly-plug-us1 esp8266 2.11.1 20220721-022206/2.11.1-3-g1f555d8-dirty sig 0x00\n"}
{"ts":1658371765.191, "level":2, "data":"shelly_update.cpp:242   Signature check: 00\n"}
{"ts":1658371765.198, "level":0, "data":"shelly_update.cpp:234   Wrong app name 'shelly-plug-us1'\n"}
{"ts":1658371765.206, "level":2, "data":"mgos_ota_core.c:1051    Update finished, result -13 (Aborted by event handler)\n"}`

just-inlewis, 2022-07-21:
Has the name changed for my plug? 
```{"name":null,"id":"shellyplugus-XXXXXXXX","mac":"XXXXXXXX","model":"SNPL-00116US","gen":2,"fw_id":"20220211-132652/plugusprod2_app-gcb4621f","ver":"plugusprod2","app":"PlugUS","auth_en":false,"auth_domain":null}```

evanpierre, 2022-08-31:
What is the state of Shelly Plug US support? yay or nay?

Attempted to install shelly-homekit today (after updating to latest stock firmware) and I get stuck with this message: "Trying to update from http://shelly.rojer.cloud/update."

Cheers and Thanks

Numbski, 2022-09-02:
I had a death in my family last year about the time a test build was made available, and I never got around to testing it.  I have five of them here, so I really should get on that.

slvr85, 2022-09-07:
There is no support for "Shelly Plus Plug US"?

RaphiePS, 2022-10-22:
First off @Numbski, so sorry for your loss. Hope you and your family have been doing well.

I've tried to compile @rojer's [commit](https://github.com/mongoose-os-apps/shelly-homekit/commit/266969e4bd2a06a2a04a3e27536877bef2f533a4) linked above, but am getting some build errors. I can wade in and try to work around them myself, but maybe this is the wrong path entirely. Anyone have suggestions?

```
/Users/raphael/Downloads/shelly-homekit/src/shelly_main.cpp: In function 'void shelly::InitApp()':
/Users/raphael/Downloads/shelly-homekit/src/shelly_main.cpp:955:58: error: too few arguments to function 'void mgos_hap_add_rpc_service_cb(HAPAccessoryServerRef*, void (*)(HAPAccessoryServerRef*), void (*)(HAPAccessoryServerRef*))'
  955 |   mgos_hap_add_rpc_service_cb(&s_server, StartHAPServerCB);
      |                                                          ^
In file included from /Users/raphael/Downloads/shelly-homekit/src/shelly_main.cpp:24:
/Users/raphael/Downloads/shelly-homekit/deps/homekit-adk/include/mgos_hap.h:126:6: note: declared here
  126 | void mgos_hap_add_rpc_service_cb(
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [/Users/raphael/Downloads/shelly-homekit/build_ShellyPlugUS1/objs/shelly_main.cpp.o] Error 1
```

gedeyenite, 2022-12-02:
> There is no support for "Shelly Plus Plug US"?

I too would like support for the Shelly Plus Plug US please. Then I can flash all my shelly stuff, hopefully, and stop using the Shelly Bridge via HOOBS. :)

gedeyenite, 2023-02-01:
> here is a new build of @rojer commit, please test it, it it works the next release will support PlugUS.
> [PlugUS.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/10556807/PlugUS.zip)

Sir- I am unsure how to install the plugin from a ZIP file in HOOBS, if that is even possible. Can it be installed through the HOOBS interface?  I would love to test it out.

RaphiePS, 2023-02-02:
Hi @timoschilling, thanks so much for your work on this. I tried the above command with your new PlugUS.zip and got a `-13 Aborted by event handler` response. Is there something else I should try?

timoschilling, 2023-02-02:
here is a new build http://rojer.me/files/shelly/misc/shelly-homekit-ShellyPlugUS.zip

RaphiePS, 2023-02-04:
Doesn't seem to be working for me, no specific error given. I'm on the Shelly 0.12 firmware, do you think that might be an issue?

timoschilling, 2023-02-05:
@RaphiePS do you have a Plug US or a Plus Plug US? If you say you have version 0.12 it sounds like a Plus Plug US, the firmware from this PR is for the old Plus US without Plus. Support for the Plus version will follow in the future.

RaphiePS, 2023-02-05:
@timoschilling Here are the results of `GET /shelly`, I think this implies a **non-** Plus plug, but I could be wrong.

 ```json
{
    "name": null,
    "id": "shellyplugus-c049ef892e60",
    "mac": "C049EF892E60",
    "model": "SNPL-00116US",
    "gen": 2,
    "fw_id": "20221206-140846/0.12.0-gafc2404",
    "ver": "0.12.0",
    "app": "PlugUS",
    "auth_en": false,
    "auth_domain": null
}
```

Note that Ii didn't come with v0.12 firmware, that was a relatively recent upgrade suggested by the Shelly Web GUI. Do you have a suggested firmware version I should downgrade to before attempting the HomeKit flash? (And if so, where I could find these older firmware files? http://archive.shelly-tools.de doesn't seem to have files for the US Plug).

timoschilling, 2023-02-05:
@RaphiePS  There three reasons why yours must be a Plus:
1. the output you posted shows `gen: 2`
2. The current version of non Plus US Plug is 1.x
3. The model of the gen 1 is [SHPLG-U1](https://shelly-api-docs.shelly.cloud/gen1/docs/coiot/v2/examples/Shelly%20Plug%20US%20SHPLG-U1]

Numbski, 2023-02-05:
I suppose I should get on trying this today, as all of mine are plug US.

Tony Shadwick

timoschilling, 2023-02-10:
@Numbski any updates?

timoschilling, 2023-02-13:
@gedeyenite If you like to test it, you could simply do it by open this url in your browser http://a.b.c.d/ota?url=http://rojer.me/files/shelly/misc/shelly-homekit-ShellyPlugUS.zip (replace a.b.c.d with the ip of your shelly)

timoschilling, 2023-02-15:
@Numbski @just-inlewis @evanpierre @ryanshuck was someone of you able to test that build? Reminder it's only for the "Plug US" not for the "Plus Plug US".

just-inlewis, 2023-02-16:
Hey @timoschilling sorry for the delay! I will keep an eye on this now to speed up your debug cycle! 
`{"ts":1676521342.232, "level":2, "data":"shelly_update.cpp:271   incoming fw signatures: 00\n"}
{"ts":1676521342.239, "level":0, "data":"shelly_update.cpp:263   Wrong app name 'shelly-plug-u1'\n"}
{"ts":1676521342.247, "level":2, "data":"mgos_ota_core.c:1051    Update finished, result -13 (Aborted by event handler)\n"}`

timoschilling, 2023-02-16:
@just-inlewis you have a Plus Plug US and not a Plug US, this build is only for the NON Plus version.

timoschilling, 2023-03-03:
Still nobody that can test it? @Numbski?

RaphiePS, 2023-03-08:
@timoschilling It looks like only the Plug Plus US is available for purchase, both on the Shelly site and Amazon. If I could get my hands on the non-Plus I'd be happy to test it, but I'm having trouble finding one and I figure others are in the same boat.

timoschilling, 2023-03-08:
@RaphiePS yes the Gen1 US Plug is out of stock.

mrala, 2023-09-19:
@timoschilling I was able to get 2.11.2 installed on my Plug US (non-plus). The device is now overheating consistently and the homekit server keeps stopping.

The device works properly after reverting back to the stock firmware.
```
{
	"type":"SHPLG-U1",
	"mac":"<snip>",
	"auth":false,
	"fw":"20230913-113516/v1.14.0-gcb84623",
	"discoverable":true,
	"num_outputs":1,
	"num_meters":1
}
```

mrala, 2023-12-20:
Are there any other logs I can provide that would be helpful?

timoschilling, 2023-12-20:
Looks like temperature measuring didn't work. We need to make a build with disabling it.

rajtotherescue, 2024-02-18:
Is there any update on firmware support for the Plus Plug US?

waffles0042, 2024-02-24:
> Is there any update on firmware support for the Plus Plug US?

humble bump. 

I would also be very interested in seeing this plug supported.