mongoose-os-apps/shelly-homekit #751: Changing HAP type on Shelly i3 is not reflected in UI until after reboot

marcus-aa, 2021-07-30:
When changing HAP type it immediately changes back to the previous setting when pressing Save. Even reloading the page still shows the previous type. But after a reboot the newly selected type is shown. Changing it again shows the same problem and requires a new reboot.

Firmware 
Version: 2.9.3
Build: 20210714-205939/2.9.3-g2f89855-master

marcus-aa, 2021-07-30:
BTW, why is it called "shellyix3" in its ID??

timoschilling, 2021-07-30:
shellyix3 is the internal name of the shelly

timoschilling, 2021-07-30:
double of #648

marcus-aa, 2021-07-30:
Similar, but not identical.  2.5 works fine when changing HAP type but does not update UI when changing main mode (or to detached input), but the mode select box IS changed.

On i3, the Hap type select box reverts by itself to old setting.

rojer, 2021-08-02:
fixed, will be in the next release.

closed 2021-08-02 by rojer