mongoose-os-apps/shelly-homekit #754: Shelly 2.5 rebooting very often

CyberMrProper, 2021-07-30:
I'm having an issue with my shelly 2.5, it's been rebooting itself very often. I have 2 more that are not having this issue (no reboot for days).

**WiFi Settings**

- RSSI: -61

**System**

- Temperature: 72°C

**Checking the log, the only error I was able to capture is the one below:**

```
1096492375 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.4.47:59890
1097490788 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.4.47:59890
1097820447 shelly_main.cpp:543     Up 1097.81, HAP 0/7/12 ns 7, RAM: 17368/38376; st 73; 4.1: c:1 mp:201.98 mt_ms:26689 cp:45.34 tp:45.34 lemd:0 lhmd:0
1098495374 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.4.47:59890
1099490640 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.4.47:59890
1100489488 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.4.47:59890
1105719708 esp_esf_buf_monitor.:75 TX is stuck!
1105724397 mgos_system.c:58        Rebooting in 1000 ms
1105733111 shelly_main.cpp:365     HAP server state: 2
1105744920 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6b3c nc 0x3fff6e2c 0/6/12
1105753508 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff732c nc 0x3fff7384 0/5/12
1105761364 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff765c nc 0x3fff6d5c 0/4/12
1105771129 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6ee4 nc 0x3fff6f1c 0/3/12
1105780752 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff8d54 nc 0x3fff6cd4 0/2/12
1105790096 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff8d7c nc 0x3fff75d4 0/1/12
1105798138 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5f8c nc 0x3fff7f34 0/0/12
1105813493 HAPPlatformTCPStrea:105 0x3fff59d4 0.0.0.0:9000 HAP connection closed, ns 0/0/12 i 1
1105833030 shelly_main.cpp:543     Up 1105.82, HAP 0/0/12 ns 0, RAM: 17136/38376; st 73; 4.1: c:1 mp:201.98 mt_ms:26689 cp:45.34 tp:45.34 lemd:0 lhmd:0
1105848893 shelly_main.cpp:365     HAP server state: 0
1105875069 mgos_sys_config.c:232   Loading conf2.json
1106013858 mgos_sys_config.c:174   Saved to conf9.json
```

Any help or hint is more than appreciated

rojer, 2021-07-30:
please try firmware from here - http://rojer.me/files/shelly/2.9.4-beta1/ 
use the firmware upload form (you'll need to use Chrome, it's been reported that it doesn't work in Safari or FF).
let me know if this solves the problem, i believe it should.

CyberMrProper, 2021-07-31:
Thanks @rojer, I've installed 2.9.4-beta1 and it seems to fix the issue. I will continue monitoring and if it says without rebooting   I will close the issue.

CyberMrProper, 2021-07-31:
I confirm that 2.9.4-beta1 fixed the issue

closed 2021-07-31 by CyberMrProper

rojer, 2021-07-31:
very good, thank you for confirming. i also fixed firmware uplaod version in safari and ff, will push 2.9.4 soon