mongoose-os-apps/shelly-homekit #756: 3-Way Switch Work Around Not Working on Multiple 2.5s

reh8388, 2021-08-01:
I cannot say whether this is a Homekit issue or a possible issue with Shelly-Homekit FW but wanted to post to see if others are experiencing this issue.

I have 3 setups using the 3 way work around shown in the wiki with the shortcut that runs to activate toggle the output based on the second switch. All 3 setups have stopped working. I have reverted the Shelly's back to OEM and complete reinstalled them into Homekit, I have removed and reinstalled the shortcut on each unit multiple times. Nothing seems to be working.

All units are running 2.9.5. I was on vacation for 2 weeks and updated as soon as I got back. Units all worked prior to leaving, however I did not check them before updating to 2.9.5.

timoschilling, 2021-08-01:
Short question are you using a iOS beta on one of your devices?

reh8388, 2021-08-01:
Great question! My Apple TVs are all running beta software (no other devices) but since those are my home hubs it is a key point I had not though of. I will do some digging into known issues with TVOS beta. Are there any known compatibility issues?

timoschilling, 2021-08-01:
That’s the problem. I run in the same issue running iOS 15 beta. https://github.com/mongoose-os-apps/shelly-homekit/issues/753

I can also reproduce it with other original HomeKit accessories. That’s why I will close this issue as non Shelly HK related.

closed 2021-08-01 by timoschilling

timoschilling, 2021-08-16:
Update: my broken case is fixed with public beta 5