mongoose-os-apps/shelly-homekit #759: Automation menu is missing

SanchoCZ, 2021-08-02:
Hi, I have several Shelly devices logged in to HomeKit.  In automation, I don't offer me the option for Arrive or Leave.  Will it work in the future? 
[image]

andyblac, 2021-08-02:
do you have location allowed for homeKit in settings ?

SanchoCZ, 2021-08-02:
Yes I have
[image]

timoschilling, 2021-08-02:
As @andyblac already said, isn’t not a issue with this firmware. HomeKit accessories didn’t have any knowledge about automatization's. That why I will close this issue. Please read the Apple documentation for help. 

But one hint: the HomeApp says it’s related to the find my iPhone settings. So maybe you need to enable find my iPhone to you that feature.

closed 2021-08-02 by timoschilling