mongoose-os-apps/shelly-homekit #760: Shelly 2.5 shutter only listens to 1 command

KrisVdSteen, 2021-08-02:
Hi,

I have a Shelly 2.5 in shutter mode with firmware 2.9.5

The problem is the Shelly only acts to 1 command and then doesn't react to any command anymore until its rebooted.
The web GUI is working fine but doesn't accept any up or down command.  Same with the Homekit connection.

Tried recalibration a couple of times, tried swapping input and output, all doesn't work.

Shutter works fine with default Shelly 2.5 firmware.

Anybody know what could be wrong here?

Thx,

KrisVdSteen, 2021-08-05:
Nobody?

lieberjott, 2021-08-05:
Can't confirm for my Shellys 2.5 (have multiple).
Updated firmware to 2.9.5.
Moving shutter down via web GUI: Works fine.
Moving shutter up via web GUI: Works fine.
Moving shutter to 60% via Homekit: Works fine.

lb-supertec, 2021-08-06:
No problems either with my shelly 2.5.
Updating from 2.9.3 to 2.9.5 went smooth.
Web GUI and home.app behave as they should.

Toyotar, 2021-10-02:
I had the same problem. Found that shutter open endstop was at same position as max open.
Shelly 2.5 measures avg power. I guess when it see a peak at avg power, it will shutdown all next commands tot protect the shutter motor. Your only way out is to reboot, and set open endstop a little lower at the shutter.
Works for me no problems anymore.

pedzp14, 2022-01-21:
my 2.5 was only reading input on chan 1 in detached on official firmware. got it to read by removing neutral loop into first L connection and only using 2nd one then L1 to sensor n back to sw1. both inputs working after that. no issues with mongoose both working after flash

masip85, 2023-03-02:
Same issue here. I've got 6 different shellys and this firmware works perfectly for 5 of them. But for one of my shellies, I can calibrate well, and work fine during 4 or 5 pushes. But at some point, it becomes unresponsive. Can't cancel continuous movement with one push. And both buttons and UI becomes completely unresponsive. It can be during a few seconds, or even indefinitely until new calibration.  

If I revert to stock firmware, It works perfectly. So I guess hardware is right. 

Another issue I have seen is that is loosing the connection quite often, even I don't see in the logs, I can see the red flag at the UI for a few seconds. 

Logs doesn't  show any error or disconnection.

```
28878402 shelly_main.cpp:443     Up 328.86, HAP 0/0/12 ns 0, RAM: 28176/21832; st 42; 4.1: c:1 mp:146.89 ip:5.00 mt_ms:14588 cp:18.42 tp:50.00 md:1 lmd:1
329658167 shelly_hap_window_c:335 WC 1: Cur pos 23.22 -> 23.90, P = 147.56
329742152 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
330458250 shelly_hap_window_c:335 WC 1: Cur pos 28.70 -> 29.39, P = 147.88
330574494 mgos_http_server.c:180  0x3fff5394 HTTP connection from 192.168.1.26:40526
330744807 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
331258274 shelly_hap_window_c:335 WC 1: Cur pos 34.19 -> 34.87, P = 147.38
331743012 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
332058293 shelly_hap_window_c:335 WC 1: Cur pos 39.67 -> 40.36, P = 147.54
332756765 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
332858381 shelly_hap_window_c:335 WC 1: Cur pos 45.31 -> 45.84, P = 147.12
333457755 shelly_hap_window_c:346 WC 1: Tgt pos 50.00 -> 49.95 (fixup)
333463305 shelly_output.cpp:63    Output 2: on -> off (moving)
333469171 shelly_hap_window_c:326 WC 1: State: moving -> stop (23 -> 24)
333713509 mgos_sys_config.c:323   Saved to conf9.json
333721346 shelly_hap_window_c:326 WC 1: State: stop -> stopping (24 -> 25)
333735577 shelly_hap_window_c:326 WC 1: State: stopping -> idle (25 -> 0)
333747726 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
334745169 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
335581354 mgos_http_server.c:180  0x3fff5394 HTTP connection from 192.168.1.26:40540
335744935 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
336380394 shelly_input.cpp:76     Input 2: change (state 1)
336386899 shelly_hap_window_c:346 WC 1: Tgt pos 49.95 -> 0.00 (ext)
336392787 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
```

timoschilling, 2023-03-02:
@masip85 can you please test how the non working Shelly behave on the location and with the window shutter of a working one. I would like to find out what exactly is the problem.

masip85, 2023-03-03:
> @masip85 can you please test how the non working Shelly behave on the location and with the window shutter of a working one. I would like to find out what exactly is the problem.

hard petition,change a shelly is a narrow hand work bad for my back. Isn't enough useful that the shelly works fine with default firmware?

masip85, 2023-12-10:
Ok. Moving it into another window, same kind of bahaviour. This is the log:
' ' ' 
112496429 shelly_debug.cpp:231    No log file, sending new entries
112609844 shelly_hap_window_c:494 WC 1: P1 = 150.530
112837859 shelly_main.cpp:443     Up 112.82, HAP 0/1/12 ns 1, RAM: 25244/18528; st 42; 4.1: c:0 mp:149.75 ip:5.00 mt_ms:14689 cp:78.13 tp:79.00 md:0 lmd:0
113409618 shelly_hap_window_c:494 WC 1: P1 = 150.646
113658011 mgos_http_server.c:180  0x3fff5fcc HTTP connection from 192.168.1.164:55922
114209823 shelly_hap_window_c:494 WC 1: P1 = 4.201
114215109 shelly_output.cpp:63    Output 2: on -> off (cal1)
114223219 shelly_hap_window_c:499 WC 1: calibration done, move_time 14487, move_power 150.299
114229334 shelly_hap_window_c:326 WC 1: State: cal1 -> postcal1 (14 -> 15)
114310953 shelly_hap_window_c:335 WC 1: Cur pos 78.13 -> 0.00, P = -1.00
114471842 mgos_sys_config.c:323   Saved to conf9.json
114480220 shelly_hap_window_c:346 WC 1: Tgt pos 79.00 -> 50.00 (postcal1)
114487240 shelly_hap_window_c:326 WC 1: State: postcal1 -> idle (15 -> 0)
114501558 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
114516646 shelly_output.cpp:63    Output 1: off -> on (move)
114523105 shelly_hap_window_c:326 WC 1: State: move -> rampup (20 -> 22)
114610231 shelly_hap_window_c:549 P = 111.65 -> 150.30
114710522 shelly_hap_window_c:549 P = 149.41 -> 150.30
114717064 shelly_hap_window_c:326 WC 1: State: rampup -> moving (22 -> 23)
115511936 shelly_hap_window_c:335 WC 1: Cur pos 4.74 -> 5.43, P = 150.37
116002028 mgos_http_server.c:180  0x3fff1bc4 HTTP connection from 192.168.1.164:49336
116312119 shelly_hap_window_c:335 WC 1: Cur pos 10.26 -> 10.95, P = 150.79
117112228 shelly_hap_window_c:335 WC 1: Cur pos 15.78 -> 16.47, P = 150.17
117913932 shelly_hap_window_c:335 WC 1: Cur pos 21.30 -> 22.01, P = 150.43
118656138 mgos_http_server.c:180  0x3fff1c3c HTTP connection from 192.168.1.164:49348
118712573 shelly_hap_window_c:335 WC 1: Cur pos 26.82 -> 27.52, P = 150.38
119511850 shelly_hap_window_c:335 WC 1: Cur pos 32.35 -> 33.04, P = 150.73
120311970 shelly_hap_window_c:335 WC 1: Cur pos 37.87 -> 38.56, P = 150.45
120838022 shelly_main.cpp:443     Up 120.82, HAP 0/1/12 ns 1, RAM: 25560/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:42.01 tp:50.00 md:1 lmd:1
121112042 shelly_hap_window_c:335 WC 1: Cur pos 43.39 -> 44.08, P = 150.74
121911887 shelly_hap_window_c:335 WC 1: Cur pos 48.91 -> 49.60, P = 150.94
121918440 shelly_hap_window_c:346 WC 1: Tgt pos 50.00 -> 49.60 (fixup)
121924006 shelly_output.cpp:63    Output 1: on -> off (moving)
121930417 shelly_hap_window_c:326 WC 1: State: moving -> stop (23 -> 24)
122165098 mgos_sys_config.c:323   Saved to conf9.json
122173176 shelly_hap_window_c:326 WC 1: State: stop -> stopping (24 -> 25)
122186866 shelly_hap_window_c:326 WC 1: State: stopping -> idle (25 -> 0)
123657476 mgos_http_server.c:180  0x3fff1c3c HTTP connection from 192.168.1.164:49364
128657446 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:50700
128837849 shelly_main.cpp:443     Up 128.82, HAP 0/1/12 ns 1, RAM: 25716/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
133655730 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:50708
136837928 shelly_main.cpp:443     Up 136.82, HAP 0/1/12 ns 1, RAM: 25896/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
138654266 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:50474
143656890 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:50486
144837535 shelly_main.cpp:443     Up 144.82, HAP 0/1/12 ns 1, RAM: 25896/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
148670965 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:38818
152837595 shelly_main.cpp:443     Up 152.82, HAP 0/1/12 ns 1, RAM: 25896/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
153657124 mgos_http_server.c:180  0x3fff40cc HTTP connection from 192.168.1.164:38824
158657063 mgos_http_server.c:180  0x3fff1c3c HTTP connection from 192.168.1.164:45940
160837629 shelly_main.cpp:443     Up 160.82, HAP 0/1/12 ns 1, RAM: 26080/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
163659279 mgos_http_server.c:180  0x3fff1c3c HTTP connection from 192.168.1.164:45956
168657410 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:46728
168837979 shelly_main.cpp:443     Up 168.82, HAP 0/1/12 ns 1, RAM: 25572/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
173658144 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:46738
176837641 shelly_main.cpp:443     Up 176.82, HAP 0/1/12 ns 1, RAM: 25896/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
178656815 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:45580
183657631 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:45592
184837868 shelly_main.cpp:443     Up 184.82, HAP 0/1/12 ns 1, RAM: 25900/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
188656722 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:44742
192837466 shelly_main.cpp:443     Up 192.82, HAP 0/1/12 ns 1, RAM: 25896/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
[... cut ...]
198659476 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:58980
200837945 shelly_main.cpp:443     Up 200.82, HAP 0/1/12 ns 1, RAM: 25900/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
203658968 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:58996
208656660 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.1.164:53542
208837990 shelly_main.cpp:443     Up 208.82, HAP 0/1/12 ns 1, RAM: 25716/18528; st 42; 4.1: c:1 mp:150.30 ip:5.00 mt_ms:14487 cp:49.60 tp:49.60 md:0 lmd:1
' ' '