mongoose-os-apps/shelly-homekit #761: Shelly 2.5: option to disable inputs for childlock

Leo-W536, 2021-08-03:
Hi, 
is it possible to create a "childlock button" to lock the phyical switch? Until now my workaround is to set "Input Mode" to "Detached" an disable both inputs, like this
[image]

Thanks in advance!

timoschilling, 2021-08-03:
His workaround breaks automatizations. 
I have the same use case and my current workflow is a Shelly one which controls the power line of the switch of a i3. 

I would like to reopen this as a feature request

rojer, 2021-08-03:
why would it break automations?

timoschilling, 2021-08-03:
Ah ok it’s make not the case for a 2.5 in with Switches, but it is for a i3 with state less inputs. The inputs disappear from HK, so the automatisation is broken

rojer, 2021-08-04:
so yes, i think the original issue was disabling inputs of a switch, and putting them to detached and then disabling the detached components is the way to go.

closed 2021-08-04 by rojer

timoschilling, 2021-08-04:
I would say no! Your way is a permanent way. I only want that from time to time to prevent the kids playing with the switches. 
With your way I need to reconfigure the switch each day. With a disable option I can simply toggle without breaking the HomeKit setup

Leo-W536, 2021-08-04:
> I would say no! Your way is a permanent way. I only want that from time to time to prevent the kids playing with the switches. 
> 
> With your way I need to reconfigure the switch each day. With a disable option I can simply toggle without breaking the HomeKit setup. 
> 
> Even the original post requested a "childlock button"

Yes, that's the reason I was thinking about. Easily disable/enable the switches from time to time.