mongoose-os-apps/shelly-homekit #763: Upgrade script not detecting devices

hughfr4nc15, 2021-08-05:
Hello,

I have a Linux VM on my laptop from where I have already updated the 20ish Shelly devices I have around the house.

For whatever reason, now, I'm not being able to update the devices because the script does not find them. I've tested and can ping a couple of them from the VM so I assume it's not a connection issue...

Here's the log:
`2021-08-05 12:24:49,102 DEBUG 847 OS: Linux
2021-08-05 12:24:49,103 DEBUG 848 app_version: 2.8.0
2021-08-05 12:24:49,103 DEBUG 849 manual_hosts:  (0)
2021-08-05 12:24:49,103 DEBUG 850 user: admin
2021-08-05 12:24:49,103 DEBUG 851 password: 
2021-08-05 12:24:49,103 DEBUG 852 action: flash
2021-08-05 12:24:49,103 DEBUG 853 mode: homekit
2021-08-05 12:24:49,103 DEBUG 854 timeout: 20
2021-08-05 12:24:49,103 DEBUG 855 info_level: 3
2021-08-05 12:24:49,103 DEBUG 856 fw_type_filter: all
2021-08-05 12:24:49,103 DEBUG 857 model_type_filter: all
2021-08-05 12:24:49,103 DEBUG 858 device_name_filter: all
2021-08-05 12:24:49,103 DEBUG 859 do_all: True
2021-08-05 12:24:49,103 DEBUG 860 dry_run: False
2021-08-05 12:24:49,103 DEBUG 861 quiet_run: False
2021-08-05 12:24:49,103 DEBUG 862 silent_run: True
2021-08-05 12:24:49,103 DEBUG 863 version: 
2021-08-05 12:24:49,103 DEBUG 864 exclude: 
2021-08-05 12:24:49,104 DEBUG 865 local_file: 
2021-08-05 12:24:49,104 DEBUG 866 variant: 
2021-08-05 12:24:49,104 DEBUG 867 verbose: 5
2021-08-05 12:24:49,104 DEBUG 868 hap_setup_code: 
2021-08-05 12:24:49,104 DEBUG 869 network_type: 
2021-08-05 12:24:49,104 DEBUG 870 ipv4_ip: 
2021-08-05 12:24:49,104 DEBUG 871 ipv4_mask: 
2021-08-05 12:24:49,104 DEBUG 872 ipv4_gw: 
2021-08-05 12:24:49,104 DEBUG 873 ipv4_dns: 
2021-08-05 12:24:49,104 DEBUG 874 log_filename: log.txt
2021-08-05 12:24:49,104 DEBUG 1559 [Device Scan] automatic scan
2021-08-05 12:24:49,104 INFO 1560 Scanning for Shelly devices...
2021-08-05 12:25:09,107 INFO 1508 
2021-08-05 12:25:09,107 INFO 1513 Devices found: 0 Upgradeable: 0 Flashed: 0
2021-08-05 12:25:09,107 INFO 1518 Log file created: log.txt`

Thanks on advance.

andyblac, 2021-08-05:
please try the v3.0 beta 
[flash-shelly-3.0.0-beta5.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6938257/flash-shelly-3.0.0-beta5.zip)

and post the log file itself. thx

hughfr4nc15, 2021-08-05:
Same result.

Log attached.

Thanks.

[log-beta.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/6938298/log-beta.txt)

andyblac, 2021-08-05:
can you run again but with `-v5` and post result thx

andyblac, 2021-08-05:
while you're doing that, are you running VLAN's ? if so make sure the computer you are running the script on is sitting in the same VLAN.

hughfr4nc15, 2021-08-05:
Hello again Andy,

Log atached. No VLANs here.

Thanks.

[log-beta.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/6938499/log-beta.txt)

andyblac, 2021-08-05:
hi sorry but I need a full debug log ie like `./flash-shelly.py -v5 --log-file=log.beta.log`

hughfr4nc15, 2021-08-05:
done, file attached but does no seem very different.

`hufranci@LinuxMint-VirtualBox:~/Scripts/shelly-homekit-master/tools$ ./flash-shelly.py -v5 --log-file=log.beta.log
load_security
OS: Linux
app_version: 3.0.0
do_all: False
dry_run: False
quiet_run: False
silent_run: False
timeout: 20
reboot: False
manual_hosts:  (0)
user: admin
password: None
action: flash
mode: homekit
info_level: 3
fw_type_filter: all
model_type_filter: all
device_name_filter: all
version: 
exclude: 
local_file: 
variant: 
force: False
verbose: 5
hap_setup_code: 
network_type: 
ipv4_ip: 
ipv4_mask: 
ipv4_gw: 
ipv4_dns: 
log_filename: log.beta.log
[Device Scan] automatic scan
Scanning for Shelly devices...

Devices found: 0 Upgradeable: 0 Flashed: 0
Log file created: log.beta.log
hufranci@LinuxMint-VirtualBox:~/Scripts/shelly-homekit-master/tools$ 
`

[log.beta.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6939225/log.beta.log)

andyblac, 2021-08-05:
hmm very strange, it should be getting something like

```
andrewblackburn@iMac tools % ./flash-shelly.py -v5
load_security
security: 192.168.3.68:
  password: '12345'
  user: admin
shelly1-FD89C2.local:
  password: '12345'
  user: admin

OS: Darwin
app_version: 3.0.0
do_all: False
dry_run: False
quiet_run: False
silent_run: False
timeout: 20
reboot: False
manual_hosts:  (0)
user: admin
password: None
action: list
mode: keep
info_level: 3
fw_type_filter: all
model_type_filter: all
device_name_filter: all
version:
exclude:
local_file:
variant:
force: False
verbose: 5
hap_setup_code:
network_type:
ipv4_ip:
ipv4_mask:
ipv4_gw:
ipv4_dns:
log_filename:
[Device Scan] automatic scan
Scanning for Shelly devices...
[Device Scan] found device: Garage-Light-SW.local, IP address: 192.168.3.21
[Device Scan] info: ServiceInfo(type='_http._tcp.local.', name='Garage-Light-SW._http._tcp.local.', addresses=[b'\xc0\xa8\x03\x15'], port=80, weight=0, priority=0, server='Garage-Light-SW.local.', properties={b'id': b'shelly1-F3BEE6', b'arch': b'esp8266', b'app': b'switch1', b'fw_version': b'2.9.0', b'fw_id': b'20210430-192906/2.9.0-g89f24b8-master', b'fw_type': b'homekit'})
[Device Scan] properties: {'id': 'shelly1-F3BEE6', 'arch': 'esp8266', 'app': 'switch1', 'fw_version': '2.9.0', 'fw_id': '20210430-192906/2.9.0-g89f24b8-master', 'fw_type': 'homekit'}
Device: Garage-Light-SW.local is Online
get_security_data
[Security] Garage-Light-SW.local
[Security] username: admin from commandline
[Security] password: None from commandline
[Device Scan] added: Garage-Light-SW.local, IP address: 192.168.3.21, FW Type: homekit, Security Enabled: False to queue
```

andyblac, 2021-08-05:
can you try `dns-sd -B` ? what does it give you?, should be something like
```
andrewblackburn@iMac tools % dns-sd -B
Browsing for _http._tcp
DATE: ---Thu 05 Aug 2021---
15:23:41.682  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
15:23:41.683  Add        3   6 local.               _http._tcp.          Kitchen-Cabinet-1
15:23:41.683  Add        3  12 local.               _http._tcp.          Kitchen-Cabinet-1
15:23:41.683  Add        3  12 local.               _http._tcp.          Bathroom-SW
15:23:41.683  Add        3   6 local.               _http._tcp.          Mums-Bedroom-SW
15:23:41.683  Add        3  12 local.               _http._tcp.          Mums-Bedroom-SW
15:23:41.683  Add        3   6 local.               _http._tcp.          Front-Garden-Light
15:23:41.683  Add        3  12 local.               _http._tcp.          Front-Garden-Light
15:23:41.683  Add        3   6 local.               _http._tcp.          Landing-Alarm-SW
15:23:41.683  Add        3  12 local.               _http._tcp.          Landing-Alarm-SW
15:23:41.683  Add        3   6 local.               _http._tcp.          Media-Room-Lamps
15:23:41.683  Add        3  12 local.               _http._tcp.          Media-Room-Lamps
15:23:41.683  Add        3   6 local.               _http._tcp.          Kitchen-Spots
15:23:41.683  Add        3  12 local.               _http._tcp.          Kitchen-Spots
15:23:41.683  Add        3   6 local.               _http._tcp.          Kitchen-Cabinet-2
15:23:41.683  Add        3  12 local.               _http._tcp.          Kitchen-Cabinet-2
15:23:41.683  Add        3   6 local.               _http._tcp.          Kitchen-Floor-Strip
15:23:41.683  Add        3  12 local.               _http._tcp.          Kitchen-Floor-Strip
15:23:41.683  Add        3   6 local.               _http._tcp.          Lounge-Light-SW
15:23:41.683  Add        3  12 local.               _http._tcp.          Lounge-Light-SW
15:23:41.683  Add        3   6 local.               _http._tcp.          shellydimmer2-40F520015A2D
15:23:41.683  Add        3  12 local.               _http._tcp.          shellydimmer2-40F520015A2D
15:23:41.683  Add        3   6 local.               _http._tcp.          shellydimmer2-40F52000B1C9
15:23:41.683  Add        3  12 local.               _http._tcp.          shellydimmer2-40F52000B1C9
15:23:41.684  Add        3   6 local.               _http._tcp.          Andys-Bedroom-SW
15:23:41.684  Add        3  12 local.               _http._tcp.          Andys-Bedroom-SW
15:23:41.684  Add        3   6 local.               _http._tcp.          Garage-Light-SW
15:23:41.684  Add        3  12 local.               _http._tcp.          Garage-Light-SW
15:23:41.684  Add        3   6 local.               _http._tcp.          shellygas-50029178B781
15:23:41.684  Add        3  12 local.               _http._tcp.          shellygas-50029178B781
15:23:41.684  Add        3   6 local.               _http._tcp.          Office-Window-Strip
15:23:41.684  Add        3  12 local.               _http._tcp.          Office-Window-Strip
15:23:41.684  Add        3   6 local.               _http._tcp.          shellydimmer-F3CD89
15:23:41.684  Add        3   6 local.               _http._tcp.          Media-Room-Light
15:23:41.684  Add        3  12 local.               _http._tcp.          shellydimmer-F3CD89
15:23:41.684  Add        3  12 local.               _http._tcp.          Media-Room-Light
15:23:41.684  Add        2   6 local.               _http._tcp.          Bathroom-SW
^C
andrewblackburn@iMac tools %
```

hughfr4nc15, 2021-08-05:
Hello Andrew,

I think I got it... Promiscuous Mode in the VM Network Settings was set to Deny... Changed to Allow All and it's working again...

Thanks for the help and sorry to bother you.

closed 2021-08-05 by hughfr4nc15

andyblac, 2021-08-05:
no problems at, glad you got it sorted.