mongoose-os-apps/shelly-homekit #764: Switch settings doesn’t save…

raphtabe, 2021-08-06:
Hello
Thanks for this great software🙂 
I’ve got the following issue 👍 
I can’t save switch settings.
The button save doesn’t work. After refresh, the values are always back to default/ empty 😢
Shelly 1PM
V 2.9.4

thanks for your help. 

[image]

Log
34051703044 shelly_debug.cpp:227    No log file, sending new entries
34057640309 shelly_main.cpp:543     Up 34057.63, HAP 0/10/12 ns 10, RAM: 19036/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34065638291 shelly_main.cpp:543     Up 34065.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34073639983 shelly_main.cpp:543     Up 34073.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34081638708 shelly_main.cpp:543     Up 34081.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34089639799 shelly_main.cpp:543     Up 34089.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34097638612 shelly_main.cpp:543     Up 34097.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34105638384 shelly_main.cpp:543     Up 34105.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34113638484 shelly_main.cpp:543     Up 34113.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34119629363 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34119662664 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34120178447 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34120225014 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34121196642 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34121232923 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34121638640 shelly_main.cpp:543     Up 34121.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34122226344 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34122296808 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34123250888 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34123298539 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34124249515 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34124298402 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34125271712 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34125303549 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34126292219 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34126358968 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34127321738 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34127348714 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34128332738 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34128385887 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34129355487 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34129380855 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34129638663 shelly_main.cpp:543     Up 34129.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34130370385 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34130413536 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34131391719 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34131427229 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34132433340 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34132459715 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34133428717 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34133459296 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34134470162 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34134493420 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34135507835 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34135534849 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34136489534 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.187:56969
34136526793 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.187:56969
34137638737 shelly_main.cpp:543     Up 34137.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34145638367 shelly_main.cpp:543     Up 34145.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34153638216 shelly_main.cpp:543     Up 34153.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34161638494 shelly_main.cpp:543     Up 34161.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34169638551 shelly_main.cpp:543     Up 34169.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34177638677 shelly_main.cpp:543     Up 34177.63, HAP 0/10/12 ns 10, RAM: 20632/39320; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
34185638406 shelly_main.cpp:543     Up

timoschilling, 2021-08-07:
Please update to 2.9.5

raphtabe, 2021-08-07:
🤦🏻‍♂️ 
ok sorry. 
2.9.5 solved my issue. 
i don’t know how i miss that upadate !
Thanks !

closed 2021-08-07 by raphtabe