mongoose-os-apps/shelly-homekit #765: Shelly 2.5 getting very hot

bastian1387, 2021-08-11:
Hey there,
i have 4 Shelly 2.5 as roller shutter and the temperature is over 80 Celcius.

[image]

[image]

[image]

[image]

doronkeren, 2021-08-16:
Same for me (fw version 2.9.5)
[image]

timoschilling, 2021-08-16:
@doronkeren 60-70 is an expensive range

rojer, 2021-08-16:
this is still ok, it's quite normal for electronics to get to 70 and up.
i am noting reports that the readings are higher for homekit fw than stock, but it could be because if a different temperature conversion curve we use for the NTC.
anyway, the actual readings are not that abnormal, it's ok.

closed 2021-08-16 by timoschilling