mongoose-os-apps/shelly-homekit #767: shelly 1pm reboots, can see IP in router and then disappears (reboots again??) and vice versa every 20 seconds

zeltak, 2021-08-13:
Hi all

i have a bunch of shelly devices and all work well

one of my shelly 1pm keeps rebooting every few seconds... I can see the IP in router and then  disappears (reboots again??) and vice versa every 20 seconds

is this a known issues, how does one try to fix this (i tried many soft and hard resets)

timoschilling, 2021-08-13:
Does it happen when you disconnect the switch from the Shelly?

rojer, 2021-08-13:
also, please check `http://a.b.c.d/debug/core` in your browser and send to riojer@rojer.me if there's anything other than "not found". if firmware crashes, this will help to tell why.

note to self (and/or @timoschilling ): show a link in UI if there's a valid core dump.

rojer, 2021-08-22:
thank you for sending in the dumps!
i think i understand what's going on.
debvice resets due to watchdog timeout and stack traces indicate that device is stuck in a loop in UART code, with output buffer full.
all the devices also show signs of UART input noise: there are log lines in the buffer about corrupt and too big incoming RPC frame, UART RX counters are high when nothing should be connected to them.
the reason this leads to devices getting stuck is the software flow control mechanism. to prevent loss of data, [XON/XOFF flow control mechanism](https://en.wikipedia.org/wiki/Software_flow_control) is enabled on UART by default. this is used to ensure reliable RPC over UART communication at decent rate without hardware flow control lines.
in this scheme, either side can ask sender to throttle by sending XOFF symbol (ASCII 19) and resume by sending XON (ASCII 17). problem is, we have noise on the line, and all sorts of symbol are being generated at random on the RX line, sometimes it happens to be XOFF, which halts transmission by the device. in core dumps i see that xoff flag is set, so device is not allowed to transmit, and when it wants to flush debug output is just spins its wheels, and eventually gets reset by watchdog.

rojer, 2021-08-23:
fix releases in 2.10.0-beta1 - #773

closed 2021-08-23 by rojer