mongoose-os-apps/shelly-homekit #768: Ability to change DNS (server) settings

fcercwer3, 2021-08-16:
It would be nice to be able to change the DNS settings, could this be implemented?

timoschilling, 2021-08-16:
Thats already possible if you use the [mongoose os cli](https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md).
`mos --port ws://<HOST>/rpc config-set wifi.sta.nameserver=8.8.8.8`

andyblac, 2021-08-16:
or script with ` --dns a.b.c.d`

rojer, 2021-12-17:
> It would be nice to be able to change the DNS settings,

y tho?
literally the only thing DNS is used for is to resolve NTP server address. i don't want to overload the UI.

rsalesas, 2023-01-16:
I'm trying to limit WAN access, what does it try to resolve?

timoschilling, 2023-01-16:
@rsalesas we use time.google.com for ntp

timoschilling, 2023-01-28:
DNS config is now added, you can load a test build here https://github.com/mongoose-os-apps/shelly-homekit/actions/runs/4028908761, at the button of the page is a releases files which includes build for each shelly.

closed 2023-02-14 by timoschilling