mongoose-os-apps/shelly-homekit #769: Can't Add Device to HomeKit Failed with OSStatus code

Steifo, 2021-08-16:
Hello,
Can't Add Device  to HomeKit (Shelly 2.5 or Shelly 1) Failed with OSStatus code -6764 (kCountErr)

Does anyone has an idea?
[image]

rojer, 2021-08-17:
can you open device log (link in the system section) and perform pairing ith the log open? then post the contents here.

Steifo, 2021-08-18:
[image]

Steifo, 2021-08-18:
I hope you mean that log

rojer, 2021-08-23:
a number of stability improvements have been made, please try 2.10 beta - #773

rojer, 2021-08-28:
hm. no idea, tbh, logs don't show anything unusual.
is it just this device or multiple?
and is it only this iDevice, or all of them?

Steifo, 2021-08-31:
My other devices are working but I am not able to add any new device to HomeKit anymore with that firmware.

timoschilling, 2023-02-02:
@Steifo is it still an issue?

closed 2023-03-07 by github-actions[bot]