mongoose-os-apps/shelly-homekit #770: Mongoose in modified shelly 2.5

moranchoj, 2021-08-17:
I have modified shelly2,5 hardware as explianed iin http://vintage.altervista.org/converting-a-shelly-2-5-into-a-2-dry-contacts-switch/?doing_wp_cron=1626700341.2935059070587158203125 to enable shelly 2.5 as a dry contact realy becuase my roller shutter motor (A-OK 45 MEL-F) requieres dry contaxt relay to send open-stop-close signal.

Modified shelly 2.5 relay together with mongoose firmware works well but as double switch not as roller shutter. Mongoose user interface always recognize two switches but do not allow to enable relay as roller shutter.

does anybody know if there is a solution?

thanks

timoschilling, 2021-08-17:
Your setup can’t work! The Shelly 2.5 doesn’t have dry contacts, only the Shelly 1 has dry contacts!

moranchoj, 2021-08-17:
With the hardware modifications explained here,

http://vintage.altervista.org/converting-a-shelly-2-5-into-a-2-dry-contacts-switch/?doing_wp_cron=1626700341.2935059070587158203125

You transform a shelly 2.5 relay in a relay with dry contacts and it works
well with mongoose formware as a 2 switches relay but not as a roller
shutter.

Does anybody have experience working with this shelly 2.5 modification?

Thanks

timoschilling, 2021-09-03:
You can use the garage door opener mode for that you want to do. The window cover mode works in a different way.
The difference is:
GDO: short impulse to open/close and a contact sensor for controlling
WC: constant power and power measuring for controlling

What you describe is the GDO mode. We already have a feature request somewhere to provide the GDO mode as WC.
I will close this issue because it is in the end a double of the that feature request.

closed 2021-09-03 by timoschilling