mongoose-os-apps/shelly-homekit #771: Shelly 2.5: Sudden Restart with switching off light

bschenker, 2021-08-18:
Since today my Shelly 2.5 with the newest Firmware 2.9.5 seems to behave very strangely: After some time (once after only 2-3 Minutes, another time after 10-15 min) the Shelly seems to turn off the attached lights and just reboots. I tried to monitor the debug log but I can't see anything special: It just suddenly stops updating with the last line just being the regular status update. 

Any idea where to start with troubleshooting?

rojer, 2021-08-18:
please check `http://a.b.c.d/debug/core` and send the contents to rojer@rojer.me if there's anything there.

bschenker, 2021-08-20:
Just sent you the respective email. I believe that issue 772 is the same as this - judging by the description.

rojer, 2021-08-23:
a number of stability improvements have been made, please try 2.10 beta - #773

closed 2021-08-23 by rojer

chstoll, 2021-09-16:
This problem still persists in 2.10. Unfortunatly there ist no core dump. Any ideas how I could help?

rojer, 2021-09-16:
when restart conicides with a power event (on or off) and leaves no core dump, that's a tell-tale sign of some kind of power rail disturbance knocking out the device. for example, an inductive load produces a current spike when de-energized. a capacitor or flyback diode is usually used to combat that.

chstoll, 2021-09-16:
I'm not sure if this is the case here. I have some LED-Panels connected to the device, and this happens randomly, sometimes after several Minutes, sometimes after several hours. But if it is something electrial, it should not be firmware-related, right? So I will flash it back to stock firmware, if the problem persists, we can be sure it is not firmware-related.