mongoose-os-apps/shelly-homekit #772: The switch switches ifself restart ( switches off)

tylichmartin, 2021-08-18:
I have Shelly 2.5 installed for two circuits (two different lights) Have you ever seen behavior that when I flash the Shelly with homekit firmware and when I turn on one of these two lights or both after some time the ligts are turned itself off and shelly for a while is not available. I am not possible to connect to he shelly via IP it seems that is disconnected or restarted. When I tried to restore to stock firmware there is everything ok..it doesn't turn itself off. So with your firmware it is not working and with stock yes. It tried also if the issue is in shelly switch and it is not I connect different shelly switch to different two circuits and it is behave the same.

timoschilling, 2021-08-18:
Sound like the overhead protection. What temperature is show in the web ui?

rojer, 2021-08-23:
a number of stability improvements have been made, please try 2.10 beta - #773

closed 2021-08-23 by rojer