mongoose-os-apps/shelly-homekit #773: Testing 2.10.0

rojer, 2021-08-23:
Ok, onto 2.10

The functionality change that merits the minor version bump is the new 1L second input mode - https://github.com/mongoose-os-apps/shelly-homekit/issues/346

Beside that, there are several stability improvements made based on analyzing the core dumps sent in by people.
https://github.com/mongoose-os-apps/shelly-homekit/issues/767 in particular should be fixed by changes to UART flow control,
and device OOMs when sending notifications to multiple connections should be fixed by aggressive LWIP memory trimming (https://github.com/mongoose-os-libs/lwip/commit/d42d197ca54338ffcb45b37c6e787faedfb01147 and https://github.com/mongoose-os-libs/lwip/commit/93ecc1107b0eed39d728cb86c4d86ba59346f849).

I encourage everyone who had stability issues before to try this firmware and report - http://rojer.me/files/shelly/2.10.0-beta1/

aspcnet, 2021-08-23:
Installed on a new 2.5 device and testing now:
1719731722 shelly_main.cpp:504     Up 1719.72, HAP 0/6/12 ns 6, RAM: 24756/42240; st 63; 4.1: c:1 mp:158.54 mt_ms:31289 cp:0.00 tp:0.00 lemd:0 lhmd:0

rojer, 2021-08-23:
@aspcnet 24K free with 6 active connections is good. can you also post a snapshot of the `/debug/info` page?

aspcnet, 2021-08-23:
Attached:
App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 6415.72
RAM: 22540 free, 16984 min free
HAP server port: 9000
HAP config number: 5
HAP connection stats: 0/6/12
HAP connections:
  192.168.66.175:58399 nc 0x3fff5874f 0x200000 io 5989 ts 0x3fff58fc rd 5988
  192.168.66.242:49991 nc 0x3fff5a0cf 0x200000 io 4407 ts 0x3fff5ffc rd 4407
  192.168.66.127:49653 nc 0x3fff56a4f 0x200000 io 4408 ts 0x3fff5fd4 rd 4407
  192.168.66.241:50086 nc 0x3fff561cf 0x200000 io 6021 ts 0x3fff4044 rd 6021
  192.168.66.250:49879 nc 0x3fff5564f 0x200000 io 6021 ts 0x3fff4774 rd 6021
  192.168.66.195:53316 nc 0x3fff408cf 0x200000 io 5572 ts 0x3fff46cc rd 5571
 Total: 6
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff46cc o 1 st 1 ts 843924
  1: s 0x3ffef328 ts 0x3fff4774 o 1 st 1 ts 394684
  2: s 0x3ffef580 ts 0x3fff4044 o 1 st 1 ts 394579
  3: s 0x3ffef7d8 ts 0x3fff5fd4 o 1 st 1 ts 2007902
  4: s 0x3ffefa30 ts 0x3fff5ffc o 1 st 1 ts 2008264
  5: s 0x3ffefc88 ts 0x3fff58fc o 1 st 1 ts 426741
 Total: 6

rojer, 2021-08-23:
16K+ min free, good. can you perform an action from the web ui (open/close curtain, for example) and check the min free number?
web action will trigger maximum number of simultaneous notifications (on all 6 connections) and will peak memory usage.
i want to see what the min free number memory goes to.

aspcnet, 2021-08-23:
While it is Opening through Web Interface:
App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 6752.43
RAM: 23252 free, 16984 min free
HAP server port: 9000
HAP config number: 5
HAP connection stats: 0/6/12
HAP connections:
  192.168.66.127:52292 nc 0x3fff53acf 0x200000 io 84 ts 0x3fff5764 rd 84
  192.168.66.175:58399 nc 0x3fff5874f 0x200000 io 6326 ts 0x3fff58fc rd 6325
  192.168.66.242:49991 nc 0x3fff5a0cf 0x200000 io 4744 ts 0x3fff5ffc rd 4744
  192.168.66.241:50086 nc 0x3fff561cf 0x200000 io 6358 ts 0x3fff4044 rd 6357
  192.168.66.250:49879 nc 0x3fff5564f 0x200000 io 6358 ts 0x3fff4774 rd 6357
  192.168.66.195:53316 nc 0x3fff408cf 0x200000 io 5909 ts 0x3fff46cc rd 5908
 Total: 6
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff46cc o 1 st 1 ts 843924
  1: s 0x3ffef328 ts 0x3fff4774 o 1 st 1 ts 394684
  2: s 0x3ffef580 ts 0x3fff4044 o 1 st 1 ts 394579
  3: s 0x3ffef7d8 ts 0x3fff5764 o 1 st 1 ts 6668698
  4: s 0x3ffefa30 ts 0x3fff5ffc o 1 st 1 ts 2008264
  5: s 0x3ffefc88 ts 0x3fff58fc o 1 st 1 ts 426741
 Total: 6

After Open action executed:

App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 6804.08
RAM: 24176 free, 16984 min free
HAP server port: 9000
HAP config number: 5
HAP connection stats: 0/6/12
HAP connections:
  192.168.66.127:52292 nc 0x3fff53acf 0x200000 io 136 ts 0x3fff5764 rd 135
  192.168.66.175:58399 nc 0x3fff5874f 0x200000 io 6378 ts 0x3fff58fc rd 6377
  192.168.66.242:49991 nc 0x3fff5a0cf 0x200000 io 4796 ts 0x3fff5ffc rd 4795
  192.168.66.241:50086 nc 0x3fff561cf 0x200000 io 6410 ts 0x3fff4044 rd 6409
  192.168.66.250:49879 nc 0x3fff5564f 0x200000 io 6410 ts 0x3fff4774 rd 6409
  192.168.66.195:53316 nc 0x3fff408cf 0x200000 io 5961 ts 0x3fff46cc rd 5960
 Total: 6
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff46cc o 1 st 1 ts 843924
  1: s 0x3ffef328 ts 0x3fff4774 o 1 st 1 ts 394684
  2: s 0x3ffef580 ts 0x3fff4044 o 1 st 1 ts 394579
  3: s 0x3ffef7d8 ts 0x3fff5764 o 1 st 1 ts 6668698
  4: s 0x3ffefa30 ts 0x3fff5ffc o 1 st 1 ts 2008264
  5: s 0x3ffefc88 ts 0x3fff58fc o 1 st 1 ts 426741
 Total: 6

After Fully Closed through Web Interface:

App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 6889.23
RAM: 22912 free, 16984 min free
HAP server port: 9000
HAP config number: 5
HAP connection stats: 0/6/12
HAP connections:
  192.168.66.127:52292 nc 0x3fff53acf 0x200000 io 221 ts 0x3fff5764 rd 220
  192.168.66.175:58399 nc 0x3fff5874f 0x200000 io 6463 ts 0x3fff58fc rd 6462
  192.168.66.242:49991 nc 0x3fff5a0cf 0x200000 io 4881 ts 0x3fff5ffc rd 4880
  192.168.66.241:50086 nc 0x3fff561cf 0x200000 io 6495 ts 0x3fff4044 rd 6494
  192.168.66.250:49879 nc 0x3fff5564f 0x200000 io 6495 ts 0x3fff4774 rd 6494
  192.168.66.195:53316 nc 0x3fff408cf 0x200000 io 6046 ts 0x3fff46cc rd 6045
 Total: 6
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff46cc o 1 st 1 ts 843924
  1: s 0x3ffef328 ts 0x3fff4774 o 1 st 1 ts 394684
  2: s 0x3ffef580 ts 0x3fff4044 o 1 st 1 ts 394579
  3: s 0x3ffef7d8 ts 0x3fff5764 o 1 st 1 ts 6668698
  4: s 0x3ffefa30 ts 0x3fff5ffc o 1 st 1 ts 2008264
  5: s 0x3ffefc88 ts 0x3fff58fc o 1 st 1 ts 426741
 Total: 6

rojer, 2021-08-23:
ok, so it didn't drop lower than previous min. very good, thank you.

rojer, 2021-08-23:
if someone has more than 6 connections and can do the same and post the numbers here, it'd be appreciated.

andyblac, 2021-08-23:
after flash, 
```
App: switch1 2.10.0-beta1 20210823-023544/2.9.5-13-g78923dc-master
Uptime: 149.77
RAM: 23720 free, 19072 min free
HAP server port: 9000
HAP config number: 43
HAP connection stats: 0/9/12
HAP connections:
  192.168.3.17:59224 nc 0x3fff587cf 0x200000 io 47 ts 0x3fff53e4 rd 47
  192.168.3.62:49172 nc 0x3fff50c4f 0x200000 io 49 ts 0x3fff5a5c rd 48
  192.168.3.11:53539 nc 0x3fff5314f 0x200000 io 12 ts 0x3fff4ea4 rd 11
  192.168.3.16:49344 nc 0x3fff4634f 0x200000 io 128 ts 0x3fff48c4 rd 127
  192.168.3.13:53320 nc 0x3fff1ab4f 0x200000 io 132 ts 0x3fff514c rd 131
  192.168.3.40:49214 nc 0x3fff5744f 0x200000 io 135 ts 0x3fff57cc rd 136
  192.168.3.55:49431 nc 0x3fff4f2cf 0x200000 io 121 ts 0x3fff4fe4 rd 121
  192.168.3.31:54128 nc 0x3fff57f4f 0x200000 io 134 ts 0x3fff55c4 rd 134
  192.168.3.29:49207 nc 0x3fff528cf 0x200000 io 135 ts 0x3fff571c rd 140
 Total: 9
HAP sessions:
  0: s 0x3ffef0b8 ts 0x3fff571c o 1 st 1 ts 14881
  1: s 0x3ffef310 ts 0x3fff55c4 o 1 st 1 ts 15760
  2: s 0x3ffef568 ts 0x3fff4fe4 o 1 st 1 ts 28748
  3: s 0x3ffef7c0 ts 0x3fff57cc o 1 st 1 ts 15169
  4: s 0x3ffefa18 ts 0x3fff514c o 1 st 1 ts 18239
  5: s 0x3ffefc70 ts 0x3fff48c4 o 1 st 1 ts 22599
  6: s 0x3ffefec8 ts 0x3fff4ea4 o 1 st 1 ts 138290
  7: s 0x3fff0120 ts 0x3fff5a5c o 1 st 1 ts 101138
  8: s 0x3fff0378 ts 0x3fff53e4 o 1 st 1 ts 102853
 Total: 9
```

web open and after actions
```
App: switch1 2.10.0-beta1 20210823-023544/2.9.5-13-g78923dc-master
Uptime: 221.46
RAM: 24088 free, 18864 min free
HAP server port: 9000
HAP config number: 43
HAP connection stats: 0/9/12
HAP connections:
  192.168.3.17:59224 nc 0x3fff587cf 0x200000 io 118 ts 0x3fff53e4 rd 118
  192.168.3.62:49172 nc 0x3fff50c4f 0x200000 io 5 ts 0x3fff5a5c rd 120
  192.168.3.11:53539 nc 0x3fff5314f 0x200000 io 83 ts 0x3fff4ea4 rd 83
  192.168.3.16:49344 nc 0x3fff4634f 0x200000 io 199 ts 0x3fff48c4 rd 199
  192.168.3.13:53320 nc 0x3fff1ab4f 0x200000 io 203 ts 0x3fff514c rd 203
  192.168.3.40:49214 nc 0x3fff5744f 0x200000 io 206 ts 0x3fff57cc rd 207
  192.168.3.55:49431 nc 0x3fff4f2cf 0x200000 io 5 ts 0x3fff4fe4 rd 192
  192.168.3.31:54128 nc 0x3fff57f4f 0x200000 io 5 ts 0x3fff55c4 rd 205
  192.168.3.29:49207 nc 0x3fff528cf 0x200000 io 206 ts 0x3fff571c rd 212
 Total: 9
HAP sessions:
  0: s 0x3ffef0b8 ts 0x3fff571c o 1 st 1 ts 14881
  1: s 0x3ffef310 ts 0x3fff55c4 o 1 st 1 ts 215820
  2: s 0x3ffef568 ts 0x3fff4fe4 o 1 st 1 ts 215819
  3: s 0x3ffef7c0 ts 0x3fff57cc o 1 st 1 ts 15169
  4: s 0x3ffefa18 ts 0x3fff514c o 1 st 1 ts 18239
  5: s 0x3ffefc70 ts 0x3fff48c4 o 1 st 1 ts 22599
  6: s 0x3ffefec8 ts 0x3fff4ea4 o 1 st 1 ts 138290
  7: s 0x3fff0120 ts 0x3fff5a5c o 1 st 1 ts 215818
  8: s 0x3fff0378 ts 0x3fff53e4 o 1 st 1 ts 102853
 Total: 9
```

rojer, 2021-08-23:
ok, very good. somewhat suprised to see switch1 with 9 connections have more ram free than 2.5 with 6, but ok.
absolute numbers look good anyway.

andyblac, 2021-08-23:
same switch1 after a few hours up
```
App: switch1 2.10.0-beta1 20210823-023544/2.9.5-13-g78923dc-master
Uptime: 28117.74
RAM: 23600 free, 18728 min free
HAP server port: 9000
HAP config number: 43
HAP connection stats: 0/8/12
HAP connections:
  192.168.3.13:54278 nc 0x3fff587cf 0x200000 io 374 ts 0x3fff5d3c rd 374
  192.168.3.17:61639 nc 0x3fff5becf 0x200000 io 1611 ts 0x3fff571c rd 1611
  192.168.3.11:56841 nc 0x3fff5b64f 0x200000 io 933 ts 0x3fff48c4 rd 932
  192.168.3.16:49939 nc 0x3fff4634f 0x200000 io 12651 ts 0x3fff4ef4 rd 12651
  192.168.3.29:49187 nc 0x3fff5304f 0x200000 io 16780 ts 0x3fff528c rd 16780
  192.168.3.40:49214 nc 0x3fff5744f 0x200000 io 28103 ts 0x3fff57cc rd 28104
  192.168.3.55:49431 nc 0x3fff4f2cf 0x200000 io 16116 ts 0x3fff4fe4 rd 16116
  192.168.3.31:54128 nc 0x3fff57f4f 0x200000 io 1023 ts 0x3fff55c4 rd 1023
 Total: 8
HAP sessions:
  0: s 0x3ffef0b8 ts 0x3fff528c o 1 st 1 ts 11337292
  1: s 0x3ffef310 ts 0x3fff55c4 o 1 st 1 ts 27094262
  2: s 0x3ffef568 ts 0x3fff4fe4 o 1 st 1 ts 12001095
  3: s 0x3ffef7c0 ts 0x3fff57cc o 1 st 1 ts 15169
  4: s 0x3ffefa18 ts 0x3fff5d3c o 1 st 1 ts 27743543
  5: s 0x3ffefc70 ts 0x3fff4ef4 o 1 st 1 ts 15466114
  6: s 0x3ffefec8 ts 0x3fff571c o 1 st 1 ts 26506442
  7: s 0x3fff0120 ts 0x3fff48c4 o 1 st 1 ts 27185042
 Total: 8
```

bschenker, 2021-08-23:
Unfortunately I only have 6 as well, but posting just in case:

57138682 shelly_main.cpp:504     Up 757.12, HAP 0/6/12 ns 6, RAM: 22780/42240; st 73; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0

App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 788.60
RAM: 21392 free, 16200 min free
HAP server port: 9000
HAP config number: 55
HAP connection stats: 0/6/12
HAP connections:
  192.168.1.117:56109 nc 0x3fff5ca4f 0x200000 io 223 ts 0x3fff5dac rd 222
  192.168.1.125:52401 nc 0x3fff5b64f 0x200000 io 83 ts 0x3fff62cc rd 384
  192.168.1.106:50530 nc 0x3fff67f4f 0x200000 io 253 ts 0x3fff62a4 rd 252
  192.168.1.127:57807 nc 0x3fff62f4f 0x200000 io 771 ts 0x3fff68c4 rd 771
  192.168.1.124:52661 nc 0x3fff676cf 0x200000 io 83 ts 0x3fff66cc rd 82
  192.168.1.122:60073 nc 0x3fff518cf 0x200000 io 248 ts 0x3fff1cac rd 247
 Total: 6
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff1cac o 1 st 1 ts 541379
  1: s 0x3ffef328 ts 0x3fff66cc o 1 st 1 ts 705685
  2: s 0x3ffef580 ts 0x3fff68c4 o 1 st 1 ts 17949
  3: s 0x3ffef7d8 ts 0x3fff62a4 o 1 st 1 ts 535798
  4: s 0x3ffefa30 ts 0x3fff62cc o 1 st 1 ts 706147
  5: s 0x3ffefc88 ts 0x3fff5dac o 1 st 1 ts 566250
 Total: 6

gregDEweb, 2021-08-23:
10248248234 shelly_main.cpp:504     Up 10248.23, HAP 0/9/12 ns 9, RAM: 21924/42240; st 57; 0.1: st:0 in_st:0 inm:1 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0

App: switch25 2.10.0-beta1 20210823-023712/2.9.5-13-g78923dc-master
Uptime: 10403.24
RAM: 21472 free, 904 min free
HAP server port: 9000
HAP config number: 105
HAP connection stats: 0/9/12
HAP connections:
  192.168.178.39:49167 nc 0x3fff4f94f 0x200000 io 840 ts 0x3fff501c rd 955
  192.168.178.28:52593 nc 0x3fff5164f 0x200000 io 840 ts 0x3fff58ec rd 2086
  192.168.178.45:49640 nc 0x3fff70a4f 0x200000 io 353 ts 0x3fff58c4 rd 352
  192.168.178.64:50500 nc 0x3fff6f2cf 0x200000 io 815 ts 0x3fff5234 rd 814
  192.168.178.27:52047 nc 0x3fff50a4f 0x200000 io 840 ts 0x3fff6e04 rd 10379
  192.168.178.24:64397 nc 0x3fff6aecf 0x200000 io 420 ts 0x3fff5ee4 rd 419
  192.168.178.36:56630 nc 0x3fff6e64f 0x200000 io 840 ts 0x3fff65d4 rd 10380
  192.168.178.85:52764 nc 0x3fff5a14f 0x200000 io 840 ts 0x3fff589c rd 10379
  192.168.178.86:60241 nc 0x3fff651cf 0x200000 io 840 ts 0x3fff6b74 rd 10380
 Total: 9
HAP sessions:
  0: s 0x3ffef0d0 ts 0x3fff58c4 o 1 st 1 ts 10050974
  1: s 0x3ffef328 ts 0x3fff6b74 o 1 st 1 ts 9563282
  2: s 0x3ffef580 ts 0x3fff58ec o 1 st 1 ts 9563276
  3: s 0x3ffef7d8 ts 0x3fff589c o 1 st 1 ts 9563281
  4: s 0x3ffefa30 ts 0x3fff65d4 o 1 st 1 ts 9563280
  5: s 0x3ffefc88 ts 0x3fff5ee4 o 1 st 1 ts 9983242
  6: s 0x3ffefee0 ts 0x3fff501c o 1 st 1 ts 9563275
  7: s 0x3fff0138 ts 0x3fff6e04 o 1 st 1 ts 9563278
  8: s 0x3fff0390 ts 0x3fff5234 o 1 st 1 ts 9588420
 Total: 9

gregDEweb, 2021-08-23:
10623202160 shelly_main.cpp:504     Up 10623.19, HAP 0/9/12 ns 9, RAM: 24396/42472; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0

App: switch1pm 2.10.0-beta1 20210823-023628/2.9.5-13-g78923dc-master
Uptime: 10616.39
RAM: 22704 free, 712 min free
HAP server port: 9000
HAP config number: 134
HAP connection stats: 0/9/12
HAP connections:
  192.168.178.39:49182 nc 0x3fff5c54f 0x200000 io 1144 ts 0x3fff5264 rd 1260
  192.168.178.64:50688 nc 0x3fff5724f 0x200000 io 1119 ts 0x3fff47ec rd 1118
  192.168.178.45:49645 nc 0x3fff5064f 0x200000 io 656 ts 0x3fff5854 rd 656
  192.168.178.85:52789 nc 0x3fff51dcf 0x200000 io 1144 ts 0x3fff4894 rd 7183
  192.168.178.24:64399 nc 0x3fff1c5cf 0x200000 io 724 ts 0x3fff6084 rd 724
  192.168.178.28:52516 nc 0x3fff58bcf 0x200000 io 1144 ts 0x3fff5894 rd 10592
  192.168.178.27:52048 nc 0x3fff63bcf 0x200000 io 1144 ts 0x3fff56fc rd 10592
  192.168.178.36:56631 nc 0x3fff52dcf 0x200000 io 1144 ts 0x3fff5b74 rd 10593
  192.168.178.86:60242 nc 0x3fff5154f 0x200000 io 1144 ts 0x3fff5124 rd 10591
 Total: 9
HAP sessions:
  0: s 0x3ffef0c0 ts 0x3fff5124 o 1 st 1 ts 9472254
  1: s 0x3ffef318 ts 0x3fff47ec o 1 st 1 ts 9497404
  2: s 0x3ffef570 ts 0x3fff5854 o 1 st 1 ts 9959932
  3: s 0x3ffef7c8 ts 0x3fff56fc o 1 st 1 ts 9472252
  4: s 0x3ffefa20 ts 0x3fff5b74 o 1 st 1 ts 9472253
  5: s 0x3ffefc78 ts 0x3fff5894 o 1 st 1 ts 9472251
  6: s 0x3ffefed0 ts 0x3fff6084 o 1 st 1 ts 9891999
  7: s 0x3fff0128 ts 0x3fff5264 o 1 st 1 ts 9472246
  8: s 0x3fff0380 ts 0x3fff4894 o 1 st 1 ts 9472248
 Total: 9

rojer, 2021-08-23:
`RAM: 21472 free, 904 min free`

`RAM: 22704 free, 712 min free`

i'm quite concerned for these two. no idea what could have caused memory to dip so low.
unfortunately, there is no history for this, but can you try the following - reboot the device, open the log and leave it running.
when min free value is updated (i.e. when memory usage reaches another peak), device prints a log line.
maybe from context i can figure out what it was doing at the time.

bmpalmeida, 2021-08-24:
Looks like noisy UART, is completely solved in 2.10. no more crashes, and no more logs about corrupted frames, and incoming rpc frame. log is clean, device is stable.

This in devices that before didn't go over 20 seconds of uptime, were having this issue: https://github.com/mongoose-os-apps/shelly-homekit/issues/767

gregDEweb, 2021-08-24:
> `RAM: 21472 free, 904 min free`
>
> `RAM: 22704 free, 712 min free`
>
> i'm quite concerned for these two. no idea what could have caused memory to dip so low.
> unfortunately, there is no history for this, [...]

They behave different now and the most Shellys are down to 0/3/12 or 0/6/12 connections. One is at 0/9/12 but values are looking ok so far. 
Somehow i‘m having network issues since i updated them. I lost connection to all devices in my HomeKit network. Not sure if this is related to the Shellys or something else. 
I restarted my network and all devices and looks ok now.

rojer, 2021-08-28:
again, of the 3 i am concerned about this one:
RAM: 23828 free, 408 min free

what could have nearly exhausted the memory? it only has 3 HAP connections, must be something else... but i have no idea what that could be. other 2.5 devices only go down to 9 and 16K, which is acceptable. and that with 3x as many connections. it's a mystery.

rojer, 2021-09-02:
i worked privately with @gregDEweb to figure out the reason of memory shortage - it ended up being a flood of UDP traffic from a hyperactive router on his network. i [made an optimization and added rate limiting](https://github.com/cesanta/mongoose/commit/311fdb8bac22c2b740be7840f1811716f659a7d2) to reduce the amount of memory that can be consumed by UDP RX buffers. thanks for you cooperation, Greg!

i built and pushed beta2, let's give it some mileage.

gregDEweb, 2021-09-02:
Thanks for all your work Rojer!

So far all devices seem to respond quicker as ever before. 
Memory on our test device after reboot with beta2 did not drop below 11k. 
I will try to collect data overnight and send it to you.

reh8388, 2021-09-06:
Running on all of my Shelly devices including 1 2.5 that had the rebooting issue previously.

Is it possible to add the 3-way switch capability being added to the 1L to the 2.5? I would appreciate getting rid of the work around without having to replace all of the 2.5 units I have.

rojer, 2021-09-06:
@reh8388 something to consider, but it's tricky for 2.5 since second input is used by the other switch. please file a separate issue for this, won't make it into 2.10.

jsiegenthaler, 2021-09-06:
Here's my experience with my Shelly1

Immediately after restart following update to 2.10.0-beta2:
`17195854 shelly_main.cpp:504     Up 17.19, HAP 0/1/12 ns 1, RAM: 27784/42512; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0`

A few minutes later:
`161195774 shelly_main.cpp:504     Up 161.19, HAP 0/4/12 ns 4, RAM: 27304/42512; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0`

After refreshing in HomeKit and turning on and off a few times:
`313196169 shelly_main.cpp:504     Up 313.19, HAP 0/3/12 ns 3, RAM: 27672/42512; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0`
`385195848 shelly_main.cpp:504     Up 385.19, HAP 0/4/12 ns 4, RAM: 27252/42512; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0`

rojer, 2021-09-06:
ok, thanks everyone, 2.10.0 is out.

closed 2021-09-06 by rojer