mongoose-os-apps/shelly-homekit #776: shelly 1 Push button function for relay status

MrSMsnoopy, 2021-08-29:
Hi,

it would be very nice if the shelly 1 could have a push button relay option as HAP function. 

I mean that the device would be activated via HomeKit app and shows the status on/activated after one press but the relay only close for a short time (for example 1 sec.) the connectors 0 and 1. After a second press the HomeKit status switched back to off/deactivated and the relay only close again for a short time (for example 1 sec.) the connectors 0 and 1.

If this were integrated, then it would be possible to control devices that change status with a short closing of the contacts. However, the status ON/OFF would remain correctly noted in HomeKit. Currently, you can this relay function with the Auto-Off but in HomeKit, the switch is shown then also OFF again. Thus, it can not be detected whether the device is on or off. 

BR,
MrSMsnoopy

timoschilling, 2021-08-29:
But how should the Shelly detect the state after a reboot?

MrSMsnoopy, 2021-08-29:
> But how should the Shelly detect the state after a reboot?

If it possible to store the last state, so he should present the last state and the relay contacts itself should be open. For sync issues could be switch in the webUI to change only the presented status.

timoschilling, 2021-08-30:
I would say that we will not implement that, because the potential of inconsistency is to high.

What’s you concrete real world use case for that?
Maybe I have a solution for you depending on your use case.

MrSMsnoopy, 2021-08-30:
I use a shelly 1 to close the connectors from a ON/OFF button from a Grundfoss control unit from a water well pump. 

The button need only a short (potential free) close connection for ON or OFF. I control also the power for the whole system and the status from the button is saved in the control unit but with hard cut from the power for the system is not healthy for the lifetime of the complete unit. So I control with an additional shelly the ON/OFF button but I can see in the HomeKit the current status from the system (only I see the power on from the whole system).

Additional if the relay contacts/ON/OFF Button are closed for the time the system is running, so the pressure control is out of order and the pump is running on max. power and stop on an over pressure condition. So that is also not a solution to hold the contacts in a closed condition.

andyblac, 2021-09-01:
this sounds like the same issue with burglar alarm thread a while ago?

MrSMsnoopy, 2021-09-02:
> this sounds like the same issue with burglar alarm thread a while ago?

I believe that will not work. If I'm not wrong the I3 need an input signal like 230V or 24-60V. I have only a potential free close contact and can't give any power over that contacts.

andyblac, 2021-09-02:
> > this sounds like the same issue with burglar alarm thread a while ago?
> 
> I believe that will not work. If I'm not wrong the I3 need an input signal like 230V or 24-60V. I have only a potential free close contact and can't give any power over that contacts.

no, what I was saying is a while a go a guy wanted to use a shelly1 to control his burglar alarm, his alarm only required a shot timed contact to activate / reactive it, but he also wanted the HK tile to show the alarm state, not thew relay state.

MrSMsnoopy, 2021-09-02:
> > > this sounds like the same issue with burglar alarm thread a while ago?
> > 
> > 
> > I believe that will not work. If I'm not wrong the I3 need an input signal like 230V or 24-60V. I have only a potential free close contact and can't give any power over that contacts.
> 
> no, what I was saying is a while a go a guy wanted to use a shelly1 to control his burglar alarm, his alarm only required a short timed contact to activate / deactivate it, but he also wanted the HK tile to show the alarm state, not the relay state.

OK, sorry I misunderstood your hint to burglar alarm topic. Any other ideas to solve this issue?

timoschilling, 2021-09-03:
Does your system have a output which is high if the system is on and low if your system is off?

MrSMsnoopy, 2021-09-03:
> Does your system have a output which is high if the system is on and low if your system is off?

The system has no direct useable output. It could be possible to use the "signal" from the power ON LED. What is your idea to use that signal?

MrSMsnoopy, 2021-09-05:
I solved that topic with an additional shelly 1. The additional shelly 1 is triggering the other shelly one on input 1 with an EDGE input mode. So can close the first shelly 1 the contacts for a short time and the other shelly 1 is presenting the status in HomeKit. 

Due to a missing output signal, it could be also show the wrong status but that could be simple synchronized with the local switch on the control unit.

closed 2021-09-05 by MrSMsnoopy