mongoose-os-apps/shelly-homekit #778: Setup code in HomeKit settings

lucaspinelli85, 2021-08-31:
Why in newest version is not possible to choose setup code, but only random? I want to able reset HomeKit server keeping older code, now is impossible because is always a newer code

rojer, 2021-08-31:
because we now have qr code and do not expect users to be entering the code manually.
while doing the UI for the QR code, i didn't think it was necessary to keep the custom code field.
so, the answer is: simplicity.
what's your use case?

timoschilling, 2021-08-31:
One use case could be if you setup the devices on an iPhone you can’t scan the code.

rojer, 2021-08-31:
true. still, given that device is only added once, it's only a minor inconvenience to enter a random code.

timoschilling, 2021-09-01:
The thing is that you need to keep the random code in mind and then switching from the safari to the home app, start the accessory adding process, waiting and then you need to type the hopefully remembered code. The problem is that you can't switch back from the home app back to safari to read the code again because the adding dialog is a modal like system UI which need to be closed before switching apps.

Maybe a solution could be that we add a like below the QR code with the url which is contained in the QR code that the device can be added via click.

lucaspinelli85, 2021-09-02:
If i setup the device, then print the qr code and write it in the device and cable it. If i need in future to reset server HomeKit, it will not be the same that i writed in the device.

rojer, 2021-09-09:
hm. i'm reluctant to add the UI for custom code back but at the same time, i see the problem with generating new code every time. maybe if the code were unique per device it would be better. problem is, the only piece of unique information per device we have is a MAC address, and that one can be considered public, and with algorithm being open one could pre-calculate setup code for a device they see on the network, which is not good.

andyblac, 2021-09-09:
@rojer what about generating a random number from the wifi password or SSID ?, where A/a = 1, B/b = 2 etc, these are not public.

rojer, 2021-09-09:
that's better but means the code changes with wifi settings. so if you rotate your wifi password (which is a good practice), the code you may have wrote down on the device is no longer valid.
of course, in practice almost nobody changes wifi password (myself included :).
what do you guys think? mix device id with wifi password to generate a (more) stable code?

rojer, 2021-09-09:
of course, i have to mention that you can use custom code through `HAP.Setup` API. invoke it with `{"code": "111-22-333"}` and bingo.
perhaps we can make it a feature of the script, for advanced use cases and leave it at that?

andyblac, 2021-09-09:
IMO I doubt anyone here os it offften enough to worry about it, as it is a PITA to change and update all your devices,

andyblac, 2021-09-09:
> of course, i have to mention that you can set a custom code through `HAP.Setup` API. invoke it with `{"code": "111-22-333"}` and bingo.
> perhaps we can make it a feature of the script, for advanced use cases and leave it at that?

and you can sett it via my script for easy change.

rojer, 2021-09-09:
oh, is that already possible? nice.

andyblac, 2021-09-09:
yup
```
andrewblackburn@iMac tools % ./flash-shelly.py -n -c 111-22-333 Office-Window-Strip.local
Looking for Shelly devices...

Host: http://Office-Window-Strip.local
Device Name: Office-Window-Strip
Model: ShellyRGBW2
Device ID: shellyrgbw2-AFAF3E
Mode: RGBW
SSID: ABFi-IoT-2.4
IP: 192.168.3.64
RSSI: -69
Up Time: 3 days, 0:15:22
HAP Connections: 0 / 8 / 12
Firmware: HomeKit 2.10.0 ✔
Would have set your HomeKit code to 111-22-333
```

timoschilling, 2021-09-10:
I think setting the code via the script didn’t fix the all problems. If you setup via a computer you can already use the QR code, if you setup with a iPhone you still fucked.

rojer, 2021-09-14:
alright, i implemented stable code based on device id, wifi ssid and password.

closed 2021-09-14 by rojer