mongoose-os-apps/shelly-homekit #779: Feature request: Add HAP Service Types for smoke and carbon monoxide sensors

roberttheiii, 2021-09-01:
I'd like to be able to connect a shelly1 to my interconnected smoke detectors. The manufacturer (kidde) sells relays to activate a 120V device in case of either a smoke or carbon monoxide alarm. I can easily put the 120V output of one of those relays into a Shelly device but I'd want the alarm to appear properly on my homekit devices. Thank you! I love this firmware.

GloWi001, 2021-09-28:
This feature could also be used to trigger critical alarms. It would also be desirable for the Shelly i3.

timoschilling, 2021-09-29:
If we implementing it, it will be available for all Shelly’s which have one or more inputs.

roberttheiii, 2022-04-29:
Amazing. Thank you.

timoschilling, 2022-04-29:
sorry mistake, smoke sensor is implemented, carbon monoxide not, #929 has implemented leak sensor.

roberttheiii, 2022-07-27:
Thank you for adding this. I have updated my Shelly 1(s) but I do not see the smoke sensor as an option. Any guidance on where I would find it? Thank you again!

timoschilling, 2023-01-31:
@roberttheiii sorry for the late response, set the "Input Mode" to "Detached", then you get a new config section for "Input 1", there you can select "HAP Type" as "Smoke sensor".

closed 2023-02-01 by timoschilling