mongoose-os-apps/shelly-homekit #780: Shelly 2.5 - Swap Inputs/Outputs doesn’t work

illadro, 2021-09-04:
I’m using Shelly 2.5 as roller shutter controller but when I try to switch inputs and outputs nothing happens.

rojer, 2021-09-04:
please elaborate - what do you expect to happen? what happens instead?

illadro, 2021-09-04:
Hi Rojer, in my case, initial situation is that OPEN and CLOSE buttons are inverted in HomeKit UI. 
If 
- Swap Inputs is ON: nothing change (buttons still inverted and physical switch works as expected)
- Swap Outputs is ON: OPEN/CLOSE buttons still inverted and the physical switch works as expected. 

Hope this can help.

rojer, 2021-09-04:
> OPEN and CLOSE buttons are inverted in HomeKit UI

what do you mean by "OPEN and CLOSE buttons are inverted in HomeKit UI"? there is only one tile for the curtain in the Home app that is used to both open and close. which "buttons" are you referring to?

"swap inputs" and "swap outputs" apply to physical inputs and outputs and do not affect homekit UI.

illadro, 2021-09-04:
When I use your web interface (device reached using its IP) open/close blue buttons appears to control the roller shutter. If I click on OPEN, the roller shutter goes down and when I click on CLOSE, it goes up. I need to invert them to match the correct directions.

illadro, 2021-09-04:
Before acting on the device swapping motor connections, I would to ask you if it's possible to swap them using the software (as Shelly Cloud has it).

rojer, 2021-09-04:
sounds like you need to swap outputs only.

rojer, 2021-09-04:
i just tested and it works as expected - with "swap outputs" enabled direction of movement in response to buttons is reversed.

illadro, 2021-09-04:
Yep. Just swapping outputs on the device the issue should be solved.

rojer, 2021-09-04:
> Yep. Just swapping outputs on the device the issue should be solved.

right. so what's the problem? it seems to be working for me.

illadro, 2021-09-04:
In my case with Swap Output enabled, open/close are still inverted. Active or not this option, the result is always the same.

rojer, 2021-09-04:
did you press "save" after changing the option? if you reload the page, does it stay activated?

illadro, 2021-09-04:
Yes it is. After click on Save and Reload, the option is active but nothing change on the device. Before flash the device, I upgraded the Shelly with the last firmware and changed outputs/inputs on the Shelly Cloud app. After that I flashed the device. Is it possible that this is causing the conflict?

rojer, 2021-09-04:
hm. i'm looking at the code and it's literally not possible.
nevertheless, let's do some testing. please open debug log in a separate tab (link in the system section), leave it open.
then go back, press open, then close with swap outputs off. then turn swap outputs on, save, then press open and close again.
after that, save the contents of the log tab, save as a [gist](https://gist.github.com/) and post a link here.

illadro, 2021-09-04:
https://gist.github.com/illadro/96244dc8c5efce5f69799ff392f761d1

rojer, 2021-09-04:
ok, i see. the problem is that even though this change requires restart of the service, it doesn't happen.
you haven't added the device to Home yet, right?

rojer, 2021-09-04:
if so, then this is a know issue, already fixed by https://github.com/mongoose-os-apps/shelly-homekit/commit/0af6d3ed9de4ea1ee0ba07f39e76c92431b3dc52 and will be included in the next release.
in the mean time, rebooting the device after making the change should do the trick.

illadro, 2021-09-04:
> ok, i see. the problem is that even though this change requires restart of the service, it doesn't happen.
> 
> you haven't added the device to Home yet, right?

Right. Just waiting to solve this issue before adding it to Home app

illadro, 2021-09-04:
> if so, then this is a known issue, already fixed by https://github.com/mongoose-os-apps/shelly-homekit/commit/0af6d3ed9de4ea1ee0ba07f39e76c92431b3dc52 and will be included in the next release.
> 
> in the mean time, rebooting the device after making the change should do the trick.

Unfortunately it doesn't solve it. Rebooting the device don't swap outputs

rojer, 2021-09-04:
enable swap outputs, save, then reboot the device. it should work.

rojer, 2021-09-04:
if not, please post the log of open/close after the reboot

illadro, 2021-09-04:
https://gist.github.com/illadro/098e4d25ec6a16df98da8cfce2b4b6cf

rojer, 2021-09-04:
please include events produced by pressing open and close, as before.

rojer, 2021-09-04:
and please include out of `http://A.B.C.D/rpc/Config.Get?key=%22wc1%22`

illadro, 2021-09-04:
https://gist.github.com/illadro/eb495a9800dc5cb4997a3a9cc8b25a14

rojer, 2021-09-04:
well, this is quite different. output swapping does work. wheeas [previously](https://gist.github.com/illadro/96244dc8c5efce5f69799ff392f761d1#file-test-L17) we had output 1 activated in response to the open command:
```
19598637011 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (RPC)
...
19598737352 shelly_output.cpp:63    Output 1: off -> on (move)
```

[now](https://gist.github.com/illadro/eb495a9800dc5cb4997a3a9cc8b25a14#file-test-L14) it's output 2:
```
28926363 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (RPC)
...
29108080 shelly_output.cpp:63    Output 2: off -> on (move)
```

however, after that we see failure of the motor to start - power consumption remains at 0, and we eventually give up and go into error state. does the motor actually move?

illadro, 2021-09-04:
After the reboot with swap outputs enabled, the motor doesn't respond. I can hear the "click"into the device but the motor doesn't work. If I disable swap outputs and reboot, the motor starts to work perfectly but open/close are still inverted.

rojer, 2021-09-04:
well, i don't know why the curtain is not moving. maybe it's because it' in the fully open position already?
try running calibration.

illadro, 2021-09-05:
Morning Rojer. Calibration has solved the issue. I really appreciate your help.

closed 2021-09-05 by timoschilling

rojer, 2021-09-05:
ok, very good. i pushed a change that resets calibration flag when outputs are swapped, thus forcing recalibration.

illadro, 2021-09-05:
> ok, very good. i pushed a change that resets calibration flag when outputs are swapped, thus forcing recalibration.

Nice! Is there a way to update all devices at the same time when a new firmware version is available?

timoschilling, 2021-09-05:
@illadro read here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script