mongoose-os-apps/shelly-homekit #785: Add "Input Mode: Deactivation" on switch

Bobbyhax, 2021-09-14:
Is it possible to add a deactivation input mode when configured as a switch? 

Just like activation, but instead of activating the output, it deactivates.

rojer, 2021-09-14:
isn't it the same as activation + inverted output?

Bobbyhax, 2021-09-14:
Yes and no. If I invert the output, it will work, but then the switch will show as off when the output is on, and that will be unlogical in my setup. 

I will use a Shelly 2.5 as a 3 speed fan controller (1 dummy switch and 2 shelly switch). When output 1 turns on, it will then turn off output 2 if it's on, by electrical connecting signal from output 2 to input 1. 
I can use automation in HomeKit to turn off the outputs, but that's a little unreliable and slow, I think.

[image]

The best thing would be to get native support for fan controller #337

rojer, 2021-09-14:
ok, would it help if we also add an option to invert status we report to homekit?

Bobbyhax, 2021-09-15:
Yes, I think that can work, but another option to consider is to make an option for allowing just 1 output to be active, 

Example: If output 1 is active, and you activate output 2, output 1 will then deactivate. If you deactivate the active output, both will be deactivated. Just like a 3 way switch.

closed 2021-10-24 by github-actions[bot]