mongoose-os-apps/shelly-homekit #787: Upgrade script says incorrect user data for --all option

steffenschmidt, 2021-09-16:
Hi,

I am using the upgrade script to flashing the shelly firmware. When running the script without mention any shelly (to update all shellys at once) I will get an error message that the credentials for my shelly aren't correct. When running the same update script for only one shelly it works fine.

Example running upgrade for all:
```
$ ./flash-shelly.py -a
Scanning for Shelly devices...

Host: http://name-of-shelly.local
Device Name: name-of-shelly
Model: Shelly25
Firmware: HomeKit 2.9.5 ➜ 2.10.0
Do you wish to flash name-of-shelly to HomeKit firmware version 2.10.0 (y/n) ? y
Downloading Firmware...
Now Flashing HomeKit 2.10.0
Host: name-of-shelly.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: my-shelly-password
```

Example when running for one shelly:
```
$ ./flash-shelly.py name-of-shelly
Looking for Shelly devices...

Host: http://name-of-shelly.local
Device Name: name-of-shelly
Model: Shelly25
Firmware: HomeKit 2.9.5 ➜ 2.10.0
Do you wish to flash name-of-shelly to HomeKit firmware version 2.10.0 (y/n) ? y
Downloading Firmware...
Now Flashing HomeKit 2.10.0
waiting for name-of-shelly to reboot.
Successfully flashed name-of-shelly to HomeKit 2.10.0

Devices found: 1 Upgradeable: 1 Flashed: 1
```

My authentication file `flash-shelly.auth.yaml` looks the following:
```
name-of-shelly.local:
  password: my-shelly-password
  user: admin
```

I didn't found any other arguments to add to perform a flash for all shellys at once.

Ok, I can write my own script which runs the firmware upgrade for every shelly but as there is the `--all` option it should work or do I miss anything?

Cheers
Steffen

andyblac, 2021-09-16:
can I have a debug log please using `--log-file` in the command line, post the file thx

andyblac, 2021-09-16:
in the mean time can you try 3.0.0 beta script 
[flash-shelly-3.0.0-beta6.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/7177101/flash-shelly-3.0.0-beta6.zip)

steffenschmidt, 2021-09-16:
Hey @andyblac 

attached you find the log file.
[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/7177626/debug.log)

Thanks for checking out. I will try the new script later and come back to you.

andyblac, 2021-09-16:
hmm it looks like you don't have have an entry in the auth yaml.,
```
2021-09-16 14:03:27,001 TRACE 703 load_security
2021-09-16 14:03:27,001 DEBUG 711 [login] name-of-shelly.local None
```

but what is also strange that auth is enabled but no pass is required, not sure how that is possible

```
2021-09-16 14:03:26,999 TRACE 226 [Device Scan] found device: name-of-shelly.local added, IP address: 10.0.0.104
2021-09-16 14:03:26,999 TRACE 227 [Device Scan] info: ServiceInfo(type='_http._tcp.local.', name='name-of-shelly._http._tcp.local.', addresses=[b'\n\x00\x00h'], port=80, weight=0, priority=0, server='name-of-shelly.local.', properties={b'id': b'shellyswitch25-AA1D3C', b'arch': b'esp8266', b'app': b'switch25', b'fw_version': b'2.9.5', b'fw_id': b'20210731-195640/2.9.5-g6108151-master', b'fw_type': b'homekit', b'failsafe': b'0', b'auth_en': b'1'}, interface_index=None)
```

andyblac, 2021-09-16:
if you are editing the debug.log as to not show info to public please email me the log file unedited to andyblac@icloud.com, it could be that the name of shelly device can not be read due eu charters or something.

steffenschmidt, 2021-09-16:
Thank you for looking deeper into it. I will mail you two log files (one with `-a -y` options and the other one with `name-of-shelly -y) and two auth files (before starting with `-a -y` and after running the script) as well. The are some interesting part which I figured out:

* when running the script for one shelly device, it will accept the credentials from the auth file
* when running with `-a -y` option it deletes the credentials from the auth file which is strange

andyblac, 2021-09-16:
ok thx, did you try v3 beta ?that is more up to date.

steffenschmidt, 2021-09-16:
That I will try now. I will come back with the results in some moments.

steffenschmidt, 2021-09-16:
With version 3.beta 6 the update runs without problems. 👍❤️

It's up to you if it is worth to put more work into version 2. If so, I am happy to assist you with testing.

andyblac, 2021-09-16:
thx for reporting, yeah version 3 has had major rewrite, it just waiting on @rojer to look over it to see if he's happy with code formatting etc, tbh not much point in trying to fix the issue in Ver2 if Ver 3.0 works ok.

But thanks for your help in testing.

steffenschmidt, 2021-09-16:
Ok understood. In that case I would close this issue and go with v3.

closed 2021-09-16 by steffenschmidt