mongoose-os-apps/shelly-homekit #788: New device

lucaspinelli85, 2021-09-16:
Is possible flash the new device?
- shelly plus 1pm
- Shelly plus 1
- Shelly 4 pro pm

andyblac, 2021-09-16:
no, these devices are not yet supported, as they use ESP32

rojer, 2021-09-16:
shelly-homekit could be easily modified to build for esp32 but there's no need: the plan is to have official homekit support for the next gen devices (in fact, i'm helping making that happen).

closed 2021-09-16 by rojer

andyblac, 2021-09-16:
let hope they do some retro fit version then, as DIN Rail version are pretty useless in UK.

rojer, 2021-09-16:
it's impossible to get esp8266-based device homekit certified by apple, so there's no chance for official homekit firmware on current generation devices.

andyblac, 2021-09-16:
> it's impossible to get esp8266-based device homekit certified by apple, so there's no chance for official homekit firmware on current generation devices.

I meant ESP32 versions of the retro fit aka Shelly1, Shelly2.5

rojer, 2021-09-16:
ah, that i don't know.