mongoose-os-apps/shelly-homekit #789: Feature request: change icon to garage door opener with others options

daitarn31974, 2021-09-19:
I would like to have the opportunity to change icon to each garage door opener I have ( actually 3 ).
One I use as gate opener, the gates moving on one side to open
The second I use is for an electric pedestrian gate
The third is for the garage door and this is the only one that has a good icon.
I mean also that's hard to recognize what is what and need to read text about to be sure.
I could even use another one for an electric lock I have and still missing the door icon.
Thank you for reply

timoschilling, 2021-09-19:
What icon do you mean, that one in the Home App?

daitarn31974, 2021-09-19:
> What icon do you mean, that one in the Home App?

exactly, in HOME APP actually there is only one icon and I can't change it. It will be very useful to have a set of different doors, garage doors and gates

timoschilling, 2021-09-19:
We have no impact on that icon, that only in the und of apple.

closed 2021-09-19 by timoschilling