mongoose-os-apps/shelly-homekit #793: Error while Updating

MrRSK, 2021-09-24:
Hi,

i'm trying to update my Shelly 2.5 from Version 2.9.1 Build 20210518-010007/2.9.1-gb7dcbef-master to 2.10.0, but i always get a NetworkError

Error uploading: TypeError: NetworkError when attempting to fetch resource.

Greetings Robert

rojer, 2021-09-25:
is there anything in the log? (link under system settings)

rhvs, 2021-09-29:
I switched off the first switch which set the temp back to 71 degrees (not sure if this is the issue though) and now I was able to start the log without crashing. Then switched on the espresso machine again :-) and below the logfile until it crashed again.

301663741 shelly_debug.cpp:231    No log file, sending new entries
301670947 mgos_http_server.c:180  0x3fff5d2c HTTP connection from 10.0.0.156:49372
304613303 shelly_main.cpp:504     Up 304.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
312613641 shelly_main.cpp:504     Up 312.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
320613739 shelly_main.cpp:504     Up 320.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
321193414 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
321937836 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
322932238 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
323939652 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
324934686 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
328613553 shelly_main.cpp:504     Up 328.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
336613698 shelly_main.cpp:504     Up 336.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
344613656 shelly_main.cpp:504     Up 344.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
352613582 shelly_main.cpp:504     Up 352.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
357127308 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
357936761 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
360613427 shelly_main.cpp:504     Up 360.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
368613567 shelly_main.cpp:504     Up 368.60, HAP 0/0/12 ns 0, RAM: 25672/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
374770897 HAPPlatformTCPStrea:121 0x3fff5c94 10.0.0.195:60507 Incoming HAP connection, ns 0/0/12
374780726 HAPPlatformTCPStrea:276 0x3fff5c94 10.0.0.195:60507 Accepted HAP connection, ns 0/1/12 ts 0x3fff5b54
376614091 shelly_main.cpp:504     Up 376.60, HAP 0/1/12 ns 1, RAM: 24716/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
384613862 shelly_main.cpp:504     Up 384.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
392613576 shelly_main.cpp:504     Up 392.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
400614185 shelly_main.cpp:504     Up 400.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
408613921 shelly_main.cpp:504     Up 408.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
416613566 shelly_main.cpp:504     Up 416.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
424613708 shelly_main.cpp:504     Up 424.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
432613636 shelly_main.cpp:504     Up 432.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
440613954 shelly_main.cpp:504     Up 440.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
448614080 shelly_main.cpp:504     Up 448.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
456613827 shelly_main.cpp:504     Up 456.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
456933101 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
457927582 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
458944061 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
459932734 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
464613550 shelly_main.cpp:504     Up 464.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0 
[... cut ...]
529934281 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
530925660 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
2893430 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
3884553 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
4880727 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
5889183 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453
6880935 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 10.0.0.156:64453

closed 2021-11-20 by github-actions[bot]