mongoose-os-apps/shelly-homekit #794: i3 toggle support

eddiebullman, 2021-09-27:
Hi 

I use a i3 with a simple on off toggle switch on the input on stock firmware but if I use the HomeKit firmware

I can no longer set this up as it only gives options for short or long press combinations on on off state.

would this be possible with the HomeKit firmware.

The reason for this is I have 2 shelly ones with PIR sensors connected to the sw inputs of them and use the i3 as and manual override to switch both Shelly ones on (i3 input switch state as on) or (i3 input switch state as off to use the PIR to trigger the Shelly ones on for a timed period)

Hope this helps what I am trying to do I would just like this native in HomeKit 

I have added a screen shot of i3

Thanks

Edd
[image]
ie

timoschilling, 2021-09-27:
In Homekit didn't support on/off on stateless devices. There are two solutions for your problem:
1. https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#stateless-switch > Input Mode > Toggle, on = single, off = double
2. https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

eddiebullman, 2021-09-27:
Hi Timo

Thanks for the help