mongoose-os-apps/shelly-homekit #795: Shelly 2.5 keeps rebooting with v2.10

rhvs, 2021-09-29:
A weird bug is currently driving me nuts (it's the shelly connected to the espresso machine, that doesn't help). It looks like the shelly is rebooting at random moments, but also when I try to access the log. Temperature is at 76 degrees, not sure if that's a threshold where the system reboots.

rhvs, 2021-09-29:
I switched off the first switch which set the temp back to 71 degrees (not sure if this is the issue though) and now I was able to start the log without crashing. Then switched on the espresso machine again :-) and below the logfile until it crashed again.

301663741 shelly_debug.cpp:231 No log file, sending new entries
301670947 mgos_http_server.c:180 0x3fff5d2c HTTP connection from 10.0.0.156:49372
304613303 shelly_main.cpp:504 Up 304.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
312613641 shelly_main.cpp:504 Up 312.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
320613739 shelly_main.cpp:504 Up 320.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
321193414 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
321937836 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
322932238 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
323939652 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
324934686 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
328613553 shelly_main.cpp:504 Up 328.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
336613698 shelly_main.cpp:504 Up 336.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
344613656 shelly_main.cpp:504 Up 344.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
352613582 shelly_main.cpp:504 Up 352.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
357127308 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
357936761 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
360613427 shelly_main.cpp:504 Up 360.60, HAP 0/0/12 ns 0, RAM: 25312/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
368613567 shelly_main.cpp:504 Up 368.60, HAP 0/0/12 ns 0, RAM: 25672/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
374770897 HAPPlatformTCPStrea:121 0x3fff5c94 10.0.0.195:60507 Incoming HAP connection, ns 0/0/12
374780726 HAPPlatformTCPStrea:276 0x3fff5c94 10.0.0.195:60507 Accepted HAP connection, ns 0/1/12 ts 0x3fff5b54
376614091 shelly_main.cpp:504 Up 376.60, HAP 0/1/12 ns 1, RAM: 24716/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
384613862 shelly_main.cpp:504 Up 384.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 72; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
392613576 shelly_main.cpp:504 Up 392.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
400614185 shelly_main.cpp:504 Up 400.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
408613921 shelly_main.cpp:504 Up 408.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
416613566 shelly_main.cpp:504 Up 416.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
424613708 shelly_main.cpp:504 Up 424.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
432613636 shelly_main.cpp:504 Up 432.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
440613954 shelly_main.cpp:504 Up 440.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
448614080 shelly_main.cpp:504 Up 448.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
456613827 shelly_main.cpp:504 Up 456.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
456933101 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
457927582 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
458944061 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
459932734 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
464613550 shelly_main.cpp:504 Up 464.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
472614038 shelly_main.cpp:504 Up 472.60, HAP 0/1/12 ns 1, RAM: 25272/42200; st 71; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 ; 0.2: st:1 in_st:0 inm:3 ininv:0; 6.2: svt:6 inv:0
[... cut ...]
528950805 mgos_mongoose.c:66 New heap free LWM: 19464
529934281 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
530925660 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
2893430 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
3884553 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
4880727 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
5889183 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453
6880935 mg_rpc.c:306 Shelly.GetInfoExt via WS_in 10.0.0.156:64453

timoschilling, 2021-09-29:
About the temperature, the threshold is between 90 and 100C

rhvs, 2021-09-30:
> About the temperature, the threshold is between 90 and 100C

OK thanks - so that can't be the issue. I'll see if I can revert to the previous fw.

Friedhofsjodla, 2021-10-02:
I have the same problem. I have two lights attached to the Shelly 2.5. If only one is turned on problem does not seem to happen at my device. I did now switch back to original firmware a few days ago and problem did not happen with the original firmware.

lieberjott, 2021-10-04:
Don't know if it helps, I have one Shelly 2.5 running firmware version 2.10.0 in roller shower mode, which means that there actions on both relay outputs multiple times a day. Uptime is now 13 days. Temperature is 63 degrees Celsius.

Friedhofsjodla, 2021-10-04:
> Don't know if it helps, I have one Shelly 2.5 running firmware version 2.10.0 in roller shower mode, which means that there actions on both relay outputs multiple times a day. Uptime is now 13 days. Temperature is 63 degrees Celsius.

I also have a couple of them running. Two of them in roller shutter mode and two with lights attached to them. The problem does only occur in the one where both lights are often turned on together for a longer period of time (about 1-2h). The others do operate without a problem.

timoschilling, 2021-10-04:
@rojer thats a issue for you

TheBigNeo, 2021-10-05:
I have the same problem
Shelly 2.5 in switch mode. Shelly restarts again and again without any recognizable pattern. It is only one that does this, the other 5 work fine. 

Model: Shelly25
Temperature: 63°C
Version:2.10.0

I have also read that one. 
The Shelly that causes problems always connects to the AP from the garden. But even if I switch off the garden AP and Shelly connects to the AP that is 3 meters away, Shelly always restarts.

https://github.com/mongoose-os-apps/shelly-homekit/issues/252

Marioishappy, 2021-10-05:
Same here , even if I have downgraded to 9.1 now. Still it reboots Occasionally

reh8388, 2021-10-06:
+2 more Shelly 2.5 units intermittently rebooting. One is far worse than the other. Happy to generate logs or anything else needed to support. The one unit typically reboots several times per hour.

rhvs, 2021-10-06:
I just noticed the original firmware maybe has the same issue, which you only notice if the power on default mode is set to "on" instead of "restore last mode". I'll try to collect logs to see if it's a hardware or software issue.

Marioishappy, 2021-10-07:
Just installed a new one without updating to last stock fw, flashed the HomeKit FW 9.1 and it's running so far.

cappadanna, 2021-10-16:
I have a Shelly 2.5 to control my garage door and it’s working great, however as soon as try to access the web login it reboots every time. I can see it disappearing in my routers list as soon as I enter it’s ip adress in the web browser

rojer, 2021-10-16:
does `/debug/core` show anything useful?

reh8388, 2021-11-02:
> I've updated to version 2.10.1 four days ago and hadn't had a single reboot since then. Temperature has also gone down by about ten degrees. So for me the problem seems to be solved - thanks 👍

Agreed 2.10.1 appears to have resolved this. I have been running for ~2 weeks now and have not seen any reboots but previously saw reboots every 10-15 minutes of the lights being.

romanbalsiger, 2021-11-03:
Same issue here with 1 out of 8 shelly 2.5 installed in my house.
The were all running on 2.9.0 with no issues. After the update on 2.10.1 the Temperature dropped on all the devices. Unfortunately one keeps rebooting. Downgrading that one solves the problem for now.

TheBigNeo, 2021-11-03:
Same here. After install 2.10.1 everything fine and also the Temp is 50°C and no more 60°C
THX

timoschilling, 2021-11-04:
sounds like we can close this issue now

closed 2021-11-04 by timoschilling

romanbalsiger, 2021-11-04:
> sounds like we can close this issue now

actually no. i stil have one that keeps rebooting under 2.10.1

timoschilling, 2021-11-05:
@romanbalsiger ok then please provide some logs from the reboot.

romanbalsiger, 2021-11-05:
it just rebootet...
so here's the 30 sec of the log:

2657833717 shelly_main.cpp:513     Up 2657.81, HAP 0/2/12 ns 2, RAM: 21396/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2658779890 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2659785679 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2660777432 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2661777286 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2662782554 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2663778936 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2664779374 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2665779298 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2665837097 shelly_main.cpp:513     Up 2665.82, HAP 0/2/12 ns 2, RAM: 21620/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2666782536 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2667782388 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2668784884 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2669782771 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2670782888 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2671785788 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2672786139 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2673797276 shelly_main.cpp:513     Up 2673.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2673809833 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2674788379 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2675787366 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2676787347 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2677791214 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2678788086 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2679792871 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2680797237 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2681797348 shelly_main.cpp:513     Up 2681.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2681809927 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2682790942 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2683791080 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2684791318 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2685792209 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2686793347 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2687796552 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2688793270 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2689797214 shelly_main.cpp:513     Up 2689.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2689810176 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2690798326 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2691797106 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2692797615 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2693797177 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2694796471 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2695797384 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2696805075 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2697797294 shelly_main.cpp:513     Up 2697.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2697810085 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2698801692 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2699801250 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2700803489 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2701802535 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2702803606 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2703803783 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2704811340 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2705797233 shelly_main.cpp:513     Up 2705.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: s [... cut ...]
2729797471 shelly_main.cpp:513     Up 2729.78, HAP 0/2/12 ns 2, RAM: 23440/42160; st 59; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:0 inm:3 ininv:0; 3.2: svt:3 inv:0 st:0 m:0 lea: -1.000
2730444416 esp_main.c:137          SDK: ip:10.10.11.12,mask:255.255.254.0,gw:10.10.10.1
2730695510 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2731693538 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360
2732694017 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.3.40:58360

timoschilling, 2021-11-05:
@rojer any idea?

romanbalsiger, 2021-11-05:
> @rojer any idea?

this concerns only 1 of 8 shelly 2.5 in my house. the rest runs smoothly on 2.10.1
all connected to the same AP, same IP Range