mongoose-os-apps/shelly-homekit #796: Switch modes for shelly 2.5 and momentary button actions

infiums, 2021-10-02:
I'm looking a way to make a button on one input work with both outputs. On original firmware there is a mode, that works so: first click turn on one channel, second click turn on second channel, then third click turns off first channel ... etc, or something like this. Can i set something like this on mongoose/shelly-homekit? Would be great to set click to operate with first channel and double click to switch second channel (output 2) and also get work long press for scene on HomeKit.

timoschilling, 2021-10-04:
In fact you can do that. You need to make a HomeKit shortcut automatisation for that. I think we will not implement that logic native in our firmware.

closed 2021-10-04 by timoschilling

infiums, 2021-10-04:
That should be in firmware, so it could work non depended on HomeKit services.

rojer, 2021-10-04:
@infiums i think you may have missed the point of this project: homekit support. it is literally in its name :)
our approach is: we want to keep the firmware a s simple as possible, anything that can reasonably be done via homekit, should be. for things that cannot, we sometimes add functionality so it's possible. and finally, we may add something that is easy enough. your request falls into that third category but does not pass the smell test - it's a bunch of logic with state tracking, and works across accessory boundaries, not really fitting well into the internal architecture. so i'm afraid it's going to be a "no".