mongoose-os-apps/shelly-homekit #797: Very flakey

base-eleven, 2021-10-03:
Love this, but perhaps due to IOS 15 or the previous updates, my Shelly 1's and 1L's are unreliable. Often slow to respond and update and sometimes completely unreachable. The web page painfully slow to load and incomplete.
Fear I'll have to try and go back to stock if it's just me, or can't be easily fixed. Thanks for all the excellent work though.

timoschilling, 2021-10-04:
Have you try to reboot your Shelly's?

base-eleven, 2021-10-04:
> Have you try to reboot your Shelly's? Do you have other Shelly's that are still working?

Yes, many times. I have five 1's and three 1L's. All come and go at will. There is no pattern that I can make out. I've heard that they are rock solid through homebridge and will have to try that next. But this was by far my preferred option. I had wondered if they IOS 15 update may have broken something, but if no-one else is having that problem, then maybe it's something else. Appreciate your reply though.

rojer, 2021-10-04:
we'll need logs to help diagnose it. please open the debug log tab on each of the devices and leave it running for a while, capturing the time when flakiness happens. then post the logs here.
also, please collect info from `http://DEVICE/debug/info`. notably - does the device reset or stay running all the time? pay attention to the uptime figure, does it reset to zero or keep increasing.

timoschilling, 2021-10-04:
This is the first time that someone reports that. I have iOS 15 since Beta 2 on iPhone, HomePod and AppleTV with around 30 Shelly’s without any issues

closed 2021-11-12 by github-actions[bot]