mongoose-os-apps/shelly-homekit #798: Shelly 2.5 Backup WiFi Access Point is enabled - how to disable?

lieberjott, 2021-10-03:
I accidentally discovered that it seems that the backup access point is enabled on one out of my ten Shelly 2.5 relays.
This means there is an unprotected WiFi connection I'm able to connect to although the Shelly is fully set up and connected to my home's WiFi.
Is there any possibility to disabled it without flashing back to stock firmware?
Prior firmware I was using was 2.9.5, today I've updated to 2.10.0 but this mode still persists. 

Thanks in advance, really appreciate.

lieberjott, 2021-10-04:
Tried to send the following command to the Shelly, which had no effect:

curl http://192.168.2.217/rpc/Config.Set -d '{"config": {"wifi": {"ap": {"enabled": false}}}, "reboot": true}'

timoschilling, 2021-10-04:
@rojer AFAK the esp can only be in AP or Client mode and not in both at the same time or?

rojer, 2021-10-04:
it is actually possible to have both AP and STA active at the same time.

rojer, 2021-10-04:
@lieberjott you are on the right track but made a small typo: you used `"enabled": false` but the key is called `"enable"`.

you can use Config.Get to get current config - `curl http://192.168.2.217/rpc/Config.Get -d '{"key": "wifi.ap"}'`.

lieberjott, 2021-10-04:
> @lieberjott you are on the right track but made a small typo: you used `"enabled": false` but the key is called `"enable"`.
> 
> you can use Config.Get to get current config - `curl http://192.168.2.217/rpc/Config.Get -d '{"key": "wifi.ap"}'`.

Yay, thanks! Was referring to https://shelly-api-docs.shelly.cloud/gen1/#settings-ap where it's called _enabled_.

closed 2021-10-04 by lieberjott