mongoose-os-apps/shelly-homekit #799: Shelly 2.5: Hissing noise and stopps working

bschenker, 2021-10-04:
Hi everyone

I'm not sure if someone can help me with this as I'm not even sure that it has something to do with Shelly-Homekit, but my Shelly 2.5 makes quite a loud hissing sound and stops working - and its also quite warm. When googeling the issue I found a video that perfectly shows the behaviour: https://vimeo.com/406197940

When I restart the shelly by removing the fuse for a few minutes to restart the Shelly it works again for a short time, but the same pattern comes back.

Any ideas how to proceed with troubleshooting? It's on the newest firmware btw.

Many thanks
Beat

timoschilling, 2021-10-04:
Sorry that video isn’t publicly watchable

Friedhofsjodla, 2021-10-05:
I think this could be related to #795. I had some strange sounds coming from my Shelly 2.5 as well and thought the hardware was broken, so I replaced it. It also rebooted suddenly as I described in the other issue. 

The replaced one does now only reboot suddenly but there are currently no strange sounds coming from it. So maybe this could be related…

bschenker, 2021-10-05:
> Sorry that video isn’t publicly watchable

That is really wired as it works fine for me without logging in. What error message do you get? 

If I would describe the sound it would be like a pipe where water runs through.

> I think this could be related to #795. I had some strange sounds coming from my Shelly 2.5 as well and thought the hardware was broken, so I replaced it. It also rebooted suddenly as I described in [...]

That does sound similar. But I'm not sure if mine is even worse as the Shelly seems to get warm during all of this.

timoschilling, 2021-10-06:
Now it works, 2 days ago I got a message like "Video not rated, you need to login to watch it".

The video definitely shows an hardware issue. Thats also the reason why I will close this issue as not firmware related.

closed 2021-10-06 by timoschilling

bschenker, 2021-10-06:
Thanks @timoschilling to confirm that this is a hardware issue.

Warps73, 2021-11-28:
Same issue for me on 3 Shelly 2.5...

AnakinSpain, 2023-01-22:
same problem with 3 shelly 2.5 after 2 years working good

tsx4k, 2023-03-11:
Same here, poor hardware used by manufacturer and 12 of Shelly 2.5 has failed: 2 dead completely, 3 had lost settings and switch to AP mode but with strange noises like the rest of them. Probably capacitors are dry after 1-2 years of working and all devices are dying :( So sad.

pnuding, 2023-06-01:
In my experience the Shelly 2.5s have a massive quality issue with the secondary side buffer capacitor of the power supply. The "ChongX" cheapo cap isn't able to withstand the warm operating environment and dries up, eventually losing the capacity to buffer the current spikes produced when the wifi broadcasts. Typical symptoms are a hissing "crssshhh" kind of noise and increasingly frequent reboots, dropping off Wifi or messing up configuration. I've had roughly 1/3rd of my Shelly 2.5 relays fail with this over the last 3 years, the ones in the warmest locations or with the most time turned on were the first to go.
All that's needed is a replacement of the (probably black) 16 volt 100 microfarad electrolyte cap located on the edge of the board, just next to one of the relays. I replace them with Panasonic FR series which are meant to have a long life in hot environments, but probably any decent brand will outlive the "ChongX"

SkywalkerPTE419, 2023-08-23:
Just to add to the issue. i had the same problem on one Shelly EM, flowed your advice and repaired my Shelly. But because was just for testing i keep the "ChongX" brand. On my case a 100microfarad 25v
[image]

petter1, 2023-10-02:
https://youtu.be/NrQVhfKCeTo?feature=shared Here is a video help for replacing the caps, which are most likely the reason for hardware defects on shelly 2.5 (maybe others as well)

pnuding, 2023-10-02:
Nice one @petter1! Where did you get the 400V caps? I never found any with the right size to fit in the space. That said so far all failures I've seen were with the secondary cap, no 400V caps have failed. Would be nice to be prepared, though 🙂

petter1, 2023-10-02:
It is not me on the video, I’m searching for compatible caps here in Switzerland right now. My two 2.5 have definitely at least one 400v broken (I see the rest of the fluid that got out and corrosion). Also in my shelly 1, there is a 8.2uF/400 definitely broken now, since it was labeled where I could not see, and now I see it but the pins are gone xD.

petter1, 2023-10-02:
Here what I have ordered:
1. 4.7uF/400v, 6.3 mm: [860021373025 Wurth Elektronik | Mouser Switzerland](https://www.mouser.ch/ProductDetail/Wurth-Elektronik/860021373025?qs=amGC7iS6iy%2FDgi%2FW1Xe8KA%3D%3D)
2. 100uF/16v, 5 mm: https://www.mouser.ch/ProductDetail/KEMET/ESL107M016AC3AA?qs=HXFqYaX1Q2xRftlnsxHD2g%3D%3D
3. (Shelly 1) 8.2uF/400v, 8mm https://www.mouser.ch/ProductDetail/Rubycon/400AX8R2MEFC8X16?qs=T3oQrply3y%252BuMrqingvaXA%3D%3D

emilof, 2023-10-20:
sS

ricargr, 2023-10-21:
Anyone with the electrical diagram of the shelly 2.5?
I need to repair quickly one of them and i used for the 16v capacitor one of low esr and 270uF. 
Now works fine but i dont want to have problema in future.

https://a.aliexpress.com/_EwpVUtv

ricargr, 2023-10-22:
The shelly technical Support response:

_Thank you for reaching out to us!
Using 270uF capacitors should not pose any issues._ 

Regards

emilof, 2023-11-05:
Thank you everyone for this thread. I just replaced the 16 volt 100 microfarad cap on my Shelly 2.5, and now the hissing noice is gone and it no longer restarts.

AnakinSpain, 2023-11-17:
> us!

Is better the same capacitors or 220uF 16v for example??

ricargr, 2023-11-17:
> > us!
> 
> Is better the same capacitors or 220uF 16v for example??

Dont think so

hazemturki, 2024-02-11:
Just to add to this issue, I started hearing hissing sounds, and then stopped working

2x Shelly 2.5

prle77, 2024-07-28:
HI guys.. will those capacitors work on dimmer 2?? I have same issue with dimmer 2 (hissing noise and dropping off network)

xbmcnut, 2024-07-29:
> HI guys.. will those capacitors work on dimmer 2?? I have same issue with dimmer 2 (hissing noise and dropping off network)

They sure will. Replace all 3 to be sure. I've done 6 of my Shellies now although mostly EM's.

FalconUK, 2025-07-30:
> > HI guys.. will those capacitors work on dimmer 2?? I have same issue with dimmer 2 (hissing noise and dropping off network)
> 
> They sure will. Replace all 3 to be sure. I've done 6 of my Shellies now although mostly EM's.

Where did you find a 4.7 µF 400 V capacitors that fit in the EM? Most are too big.

xbmcnut, 2025-07-30:
N.B In the original Shelly dimmer, this also goes bad https://nz.rs-online.com/web/p/aluminium-capacitors/7075830

FalconUK, 2025-07-31:
> I haven’t found the right size 400V, no. But so far I have never had a Shelly fail because of those, it was always just the low voltage one

I have an EM here with an auto-disassembled 4.7 µF 400 V cap. The can was found on the other side of the enclosure, next to the relay. 

[image]