mongoose-os-apps/shelly-homekit #800: Plus series support

ARumberg, 2021-10-04:
Hi,

Will the new Shelly Plus series be supported in the near future?

Thanks.

timoschilling, 2021-10-04:
Not by this firmware, but Shelly announces HomeKit support for stock firmware

rojer, 2021-10-04:
yes, there will be official homekit support and yours truly is helping make it happen. stay tuned.
but meeting all the official requirements and passing apple certification will not be easy, so it will take some time.

LMMAMP, 2021-10-30:
For one person that is not very experienced in this coding themes, will there be a solution for shelly plus 1 easy to use like the ones you did for shelly 2.5 and others?
Thanks!

timoschilling, 2021-11-01:
I will try to build a online flash tool

LMMAMP, 2021-11-01:
> I will try to build a online flash tool

Great! Any reason to not having included the sensors in the past (motion sensor and door/window 2 sensor)?

Thanks!

mkuerten, 2021-11-09:
I'm willing to give the alpha a go. I included the Shelly 1 Plus with the Shelly Beta app. Then I copied the command above in the terminal window of Mac OS. But I get this: -bash: $: command not found.

rojer, 2021-11-09:
> I'm willing to give the alpha a go. I included the Shelly 1 Plus with the Shelly Beta app. Then I copied the command above in the terminal window of Mac OS. But I get this: -bash: $: command not found.

omit the `$`, it's there just to illustrate command prompt. i think in modern mac os with zsh it's actually `%`.

mkuerten, 2021-11-09:
Sorry, stupid novice error. Probably one more: The first command runs well, but with the second ("curl -F file=@shelly-homekit-ShellyPlus1.zip http://192.168.1.57/update") I get "curl: (26) Failed to open/read local data from file/application". 

Thanks for your help!

timoschilling, 2021-11-09:
have you downloaded the firmware file before?
`curl -O http://rojer.me/files/shelly/2.11.0-alpha1/shelly-homekit-ShellyPlus1.zip`
you need to run both commands in the same directory.

mkuerten, 2021-11-09:
First I ran this:
curl -O http://rojer.me/files/shelly/2.11.0-alpha1/shelly-homekit-ShellyPlus1PM.zip
I got this:
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1606k  100 1606k    0     0  3516k      0 --:--:-- --:--:-- --:--:-- 3516k
Then I ran this:
curl -F file=@shelly-homekit-ShellyPlus1.zip http://192.168.1.57/update
They were run from the exact same prompt.

rojer, 2021-11-09:
can you run `ls -l shelly-homekit-ShellyPlus1.zip` in the same session? this is weird.

mkuerten, 2021-11-09:
I ran both again in a new terminal window and now it worked. Thanks for your help!!!

rojer, 2021-11-09:
great. let me know what your experience with the firmware is.

mkuerten, 2021-11-09:
It's brilliant! I have set it up in the home app and there it works like any other Shelly so far. Thanks so much again!

Naveto, 2021-11-09:
I have installed seven Shelly 1 plus and they work perfectly.
The only problem I find is that if you leave AP mode enabled, there is no way to disable it from the shelly-homekit interface.

Great job.

LMMAMP, 2021-11-10:
Hello, 
I really don't know much about this and i have tried to go with the plus 1 but didn't succeed...
1) I went to the windows powershell (i think i don't have the terminal )
2) Typed  " % curl -O http://rojer.me/files/shelly/2.11.0-alpha1/shelly-homekit-ShellyPlus1.zip " and it gave me this error:
"ForEach-Object : Parameter cannot be processed because the parameter name 'O' is ambiguous. Possible matches include:
-OutVariable -OutBuffer.
At line:1 char:8
+ % curl -O http://rojer.me/files/shelly/2.11.0-alpha1/shelly-homekit-S ...
+        ~~
    + CategoryInfo          : InvalidArgument: (:) [ForEach-Object], ParameterBindingException
    + FullyQualifiedErrorId : AmbiguousParameter,Microsoft.PowerShell.Commands.ForEachObjectCommand"

Is it possible to guide me a little bit here?

Thanks

LMMAMP, 2021-11-10:
I have dropped the "%" .
Now the message is the following:
"cmdlet Invoke-WebRequest at command pipeline position 1
Supply values for the following parameters:
Uri:  "

timoschilling, 2021-11-10:
@LMMAMP are you using Windows? The commands are only for Linux and Mac. 

Sorry I don’t habe the knowledge for Windows to help you

LMMAMP, 2021-11-10:
> @LMMAMP are you using Windows? The commands are only for Linux and Mac.
> 
> Sorry I don’t habe the knowledge for Windows to help you

Hello,

yes , I am using windows . Don’t have a Mac and don’t have Linux installed anywhere .

but I used the development for the Shelly 1 and the 2.5 and worked great . 

I am only having problems  in this new series app.

thanks

ARumberg, 2021-11-10:
You can download curl for windows, and run the same commands in CMD.

LMMAMP, 2021-11-10:
> You can download curl for windows, and run the same commands in CMD.
> […](#)

Thanks! I downloaded and succeed in moving 7 shelly plus1 pm to the HomeKit. Looks good!

LMMAMP, 2021-11-10:
> drop the `%`, it just indicates command prompt.

No working . Moved 7 shelly plus 1. Looks good . Thanks !

LMMAMP, 2021-11-10:
> > drop the `%`, it just indicates command prompt.
> 
> No working . Moved 7 shelly plus 1. Looks good . Thanks !

Is something projected for the sensors of Shelly ( door / window 2  and motion )? Thanks again !

timoschilling, 2021-11-11:
see #124 & #693

LMMAMP, 2021-11-11:
> @LMMAMP see #124 & #693

Thanks. If i understood right, not in the “to do” list anymore as technically it is very difficult or even impossible , right?

tobification, 2021-11-11:
I also updated 1 Shelly Plus and integrated it into HomeKit successfully. 
One thing I identified, that the "Auto off" function I'm not able to save. 
Anyone else the same issue?

rojer, 2021-11-11:
> > @LMMAMP see #124 & #693
> 
> Thanks. If i understood right, not in the “to do” list anymore as technically it is very difficult or even impossible , right?

correct. not impossible but don't hold your breath.

rojer, 2021-12-17:
fixed in 2.11.0-alpha2, see #868

closed 2021-12-17 by rojer

frankpeters1001, 2021-12-27:
Hi,

I tried to update a ShellyPlus1PM, but I got some errors. First, let me explain what I did so that you have a better understanding of whats going on:
I downloaded the file, copied it on the d drive and ran th command prompt as administrator in the directory where I copied the zip file to. I did not get any error at this time. The command prompt showed one device udpated. 
When I tried to connect with Firefox, I got the following messages:
- Failed to apply settings. No handler for shelly.getdeviceinfo
- Failed to apply settings. No handler for shelly.getconfig
- Failed ..... No handler for webhook.listsupported
- Failed...... No handler for schedule.list
- Failed ..... No handler for scipt.list

I do not see anything else in the browser.

Any ideas of what I am doing wrong here?

You help would be appreciated.

rojer, 2021-12-27:
@frankpeters1001 this looks like old interface cached by firefox trying to talk to the new firmware.
try reloading the page and closing and re-opening the browser.

frankpeters1001, 2021-12-27:
Thanks for the quick response. It worked. I will flash the other 3 tomorrow or the day after and keep you posted about any issues. Thanks a lot for your help.

frankpeters1001, 2021-12-27:
Thanks. That was easy. I will flash the other 3 tomorrow or the day after and I will let you know if any issues occur. Thanks a lot for your help.

frankpeters1001, 2021-12-29:
Hi,

Just wondering if/when the software supports the measurement of the current. 

Looking forward to your response.

rojer, 2021-12-30:
> Hi,
> 
> Just wondering if/when the software supports the measurement of the current.
> 
> Looking forward to your response.

you want current value reported along with power?

frankpeters1001, 2021-12-30:
Hi,

Sorry, I meant power measurement.

rojer, 2021-12-31:
in that case, i'm not sure what you mean because we already support it - 1PM and Plus1PM both do.

frankpeters1001, 2021-12-31:
Hi,

How do I see the power consumption in the HomeKit app? Perhaps, I just do not know how because I am new with this kinda stuff. 

Your help is really appreciated. 

Best wishes

rojer, 2021-12-31:
standard home app does not show power consumption, there's no way.
you can see it on the web interface and we also export charactestics used by the Eve app, so you can see it there too.

nikwest, 2021-12-31:
You cannot see power consumption in the Home App. You need a 3rd Party app like https://apps.apple.com/de/app/eve-for-homekit/id917695792?l=en <https://apps.apple.com/de/app/eve-for-homekit/id917695792?l=en>

Greetings

Dominik

frankpeters1001, 2022-01-01:
Hi Dominik,

Thanks for your advice. I downloaded the app. Can I see power consumption in eve or web interface, even though, I updated the software so that I can see the Shelly in the HomeKit app?

timoschilling, 2022-01-01:
If you updated the software you can see the power consumption in the Home App, Eve App and the Web UI

frankpeters1001, 2022-01-01:
Hi,

The web UI shows the power consumption. The eve app shows the power, but not the consumption and I do not find it inside the home app at all. 

Am I doing something wrong or do I look at the wrong place?

[image]

[image]

fred80, 2022-01-16:
Hello,
I've purchased a Shelly Plus 1 and installed alpha8. It works great with HomeKit! 
However I cannot get the notifications to work with it. I would like iOS to notify me, if the contact has been closed (I have a baby that doesn't sleep that well so I turn of the chime. With Shelly on the door bell I can get notified in my mobile if someone rings the door). Is this an option or does it still need to be developed?

rojer, 2022-01-16:
can you clarify what you mean by "contact has been closed" - are you referring to an external contact switch connected to shelly's input?

fred80, 2022-01-16:
> can you clarify what you mean by "contact has been closed" - are you referring to an external contact switch connected to shelly's input?

Hi Rojer,

I meant the input in the Shelly. I have a door bell connected to it. So if someone rings the door bell it closed the contact and actuates the input in the Shelly.

rojer, 2022-01-16:
then you need to first set switch input mode to "detached", which will create a separate service for the input. then you can attach automations to the input. might want to select mode for the input to be "contact sensor" too.

fred80, 2022-01-16:
> then you need to first set switch input mode to "detached", which will create a separate service for the input. then you can attach automations to the input. might want to select mode for the input to be "contact sensor" too.

Works great! Thank you very much. :)

pedzp14, 2022-01-24:
2 curl commands and a great automated interface surely theres no need to remove 2nd curl. software works great as usuall no issues.

fred80, 2022-01-27:
Hello, I've found a bug: if you select in HAP Type "Doorbell" (simple mistake from my part) then you cannot change to anything else. If you try to change it, it reverts back to "Doorbell" after 1 or 2 seconds. To get the Shelly back working, one has to revert to factory setting and do the curl commands again.

timoschilling, 2022-01-27:
please read the first post of #868

MokkaSchnalle, 2022-02-18:
There is now an easier way to flash the Plus model.
Just upgrade to the original 0.10 Beta firmware which enables the /ota endpoint again.

Then you can flash the firmware in the same way as on the older models.

1yoda, 2022-02-21:
> Just upgrade to the original 0.10 Beta firmware which enables the /ota endpoint again.

How do I do this? Where can I find and how can I flash the "original 0.10 Beta"?

MokkaSchnalle, 2022-02-21:
Just click the update beta button on the webinterface (firmware update tab)

timoschilling, 2022-03-29:
@LoicDelorme they says it in plus/pro presentation video and you will finde some comments from the CEO about it on Facebook

jsurpless, 2022-05-08:
I'd like to start by saying that this is all very impressive, nice work!

I tried this out yesterday and then restored back to the OEM firmware.  The device seemed to work a bit finicky until I reset back to factory (for instance, Shelly Cloud app would often show as OFFLINE until I closed and re-opened the app).  

Did I do something wrong?

rojer, 2022-05-08:
> The device seemed to work a bit finicky until I reset back to factory (for instance, Shelly Cloud app would often show as OFFLINE until I closed and re-opened the app).
> Did I do something wrong?

we are not aware of issues after switching back, and "a bit finicky" and "sometimes offline" is not specific enough. we try to wipe all the state introduced by HK firmware but it's always a good idea to perform factory reset after changing.

> Also, any "harm" in configuring the device in HomeKit and then restoring to Shelly firmware without removing it from HomeKit? I see that HomeKit reports device unreachable but will it be pinging the device (perhaps why I was having issues)?

there shouldn't be any harm. as long as the device stops resolving on bonjour, HK should not be contacting it anymore.

jsurpless, 2022-05-08:
> we are not aware of issues after switching back, and "a bit finicky" and "sometimes offline" is not specific enough. we try to wipe all the state introduced by HK firmware but it's always a good [...]

Hi Rojer

Thanks for the reply.  

By 'sometimes offline', I was trying to say that the module would say 'OFFLINE' under its name and tapping it would bring up 'EDIT DEVICE', as opposed to being able to configure it.  The only way that I could get it to work properly would be to quit the app on my iPhone and re-open.

LMMAMP, 2022-07-28:
Hello,

Do you have already ( or scheduled) the door 2  sensor integration to apple homekit firmware?

Thanks
Luis Pereira 

Enviado do meu iPhone

rojer, 2022-07-28:
no, not yet