mongoose-os-apps/shelly-homekit #808: URL (Turn ON or OFF) for Shelly RGBW2

StevenBeck93, 2021-10-15:
Hello,

i tried to turn the Shelly RGBW2 with URL Code on or off, but this Url doesnt work: 

/rpc/Shelly.SetState?id=1&type=0&state=%7b"state"%3atrue%7d

with a Shelly 1, the URL is working.

Please help me, to find the correct URL for a RGBW2.

Thanks

StevenBeck93, 2021-10-15:
Perfect, thank you very much! It is working :-)

closed 2021-10-15 by StevenBeck93

manueltuescher, 2023-03-04:
how can i take this HAP types and paste it into my RGBW?? i dont get it.. pls help

timoschilling, 2023-03-04:
You just need to use this url to turn on your rgbw2 http://a.b.c.d/rpc/Shelly.SetState?id=1&type=11&state=%7b"state"%3atrue%7d

manueltuescher, 2023-03-04:
Hi Timo

Thank you for your answer.. this URL dont work. 

What can i do??

thanks for your help 😊

Mit freundlichen Grüssen

Manuel Tüscher
Projektleiter Elektroplanung / IT
Berufsbildner

 ***@***.***> ***@***.***
079 297 14 53
 <https://www.ast-elektro.ch/> www.ast-elektro.ch
 <https://ast-elektro.ch/> 

 <https://g.page/r/CRTr25MkaZ1ZEB0/review> 

P Bitte denken Sie an die Umwelt, bevor Sie dieses Mail drucken

manueltuescher, 2023-03-04:
i get this message if i try this url.. 

[image]

i only use the W output. R, G and B arent in use..

timoschilling, 2023-03-04:
Are you using RGB+W mode or White Mode?

manueltuescher, 2023-03-04:
i use "White Mode"

danricho, 2023-03-04:
Once a Shelly device is configured and working, I find it helpful to get the information needed to adjust it via the API (such as type) using the get extended info endpoint:

`rpc/Shelly.GetInfoExt`

timoschilling, 2023-03-04:
@manueltuescher then it should be id=4 in the url I posted above. 

@danricho I plan to show api urls in the UI